InfiniTimeOrg/InfiniTime #947: Watch softlocks upon charge completion

g1t3m, 2022-01-20:
I've been having this issue a few months after receiving my Pinetime. Everytime I would charge the watch to 100% battery, the screen would turn black though still remain on, indicating the watch has softlocked and does not usually return to normal operation until the watch has been connected to the charger after a week or so. The battery is also not completely depleted before it reactivates.

Steps to recreate issue:
Connect Pinetime watch to charger.
Connect charger to outlet or battery pack.
Leave watch to charge for a few hours.
Check watch.
Watch will no longer display watchface or any other screen when tapping or pressing the side button, but will still be backlit indicating watch is still on.
Wait a few days to a week.
Watch will soft reset upon reconnection to charger connected to outlet.

Please let me know if this issue can be resolved.

Thank you.

Riksu9000, 2022-01-20:
Have you updated to the latest version of InfiniTime (1.8.0)? If not, please update it.

Is the backlight still on after a week? I would imagine the battery actually **has** depleted, and that's why it restarts when connecting to a charger. When the watch wakes up, it probably doesn't show 0%, but this doesn't mean that it isn't empty. The displayed percentage is higher that the real charge while the watch is plugged into a charger.

Normally you can hold the button down ~7s to force restart the watch. Can this also recover the watch without having to wait for up to a week?

g1t3m, 2022-01-20:
Hello,

I updated to the latest firmware from Github last nighr. I checked back this
morning and was able to reboot the watch by holding down the button per your
instructions.
The battery icon was shown as depleted. I held down the power button to turn off
the watch, and now the watch will not power on again, likely because the battery
has been depleted. I'll connect it to my charger to see if that resolves the
issue, unless there's something else I need to do before connecting to the
charger? Will keep you updated if anything changes.

Thank you!

Have you updated to the latest version of InfiniTime (1.8.0)? If not, please
update it.

Is the backlight still on after a week? I would imagine the battery actually has 
depleted, and that's why it restarts when connecting to a charger. When the
watch wakes up, it probably doesn't show 0%, but this doesn't mean that it isn't
empty. The displayed percentage is higher that the real charge while the watch
is plugged into a charger.

Normally you can hold the button down ~7s to force restart the watch. Can this
also recover the watch without having to wait for up to a week?

—
[https://github.com/InfiniTimeOrg/InfiniTime/issues/947#issuecomment-1017698948]
or Android

Sent with Criptext secure email

g1t3m, 2022-01-24:
Update: I've attempted several times to charge the watch up to and past 90%. All these attempts caused the watch screen to go blank, effectively softlocking the watch until it reached about 75% battery remaining. Is this a battery issue?

JF002, 2022-01-25:
This is a very strange issue, as the charging of the battery is completely done by a dedicated chip, without the intervention of the firmware. InfiniTime only checks that status of a GPIO to determine if the battery is charging and measure the voltage level on another one to estimate the battery level.
So, at first sight, there's no reason for the firmware to freeze when the battery is charged > 90 % buuuuut you're showing us that this is not completely true...

g1t3m, 2022-01-27:
Would temperature and the location the watch is kept in have any impact? Cosmetically, the watch is fine, but I'm not sure about the battery.

alwayslurking, 2022-01-27:
I think I also have this problem. Immediately after charging my watch won't display anything, but will still (sometimes) connect to gadgetbridge and show the backlight. It eventually starts behaving normally a while (hours, sometimes days) after charging, but I can't see any pattern as to when or why that happens.

Riksu9000, 2022-06-12:
I'll assume this is a hardware issue and not an InfiniTime issue.

closed 2022-06-12 by Riksu9000