InfiniTimeOrg/InfiniTime #950: Multiple Alarms

bokf, 2022-01-22:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Have multiple alarms running, for example 7am and 15pm daily

### Description

It would be nice to have the possibility to have multiple alarms running at the same time. For example one recurring alarm from Mo-Fr at 7am and then a second alarm daily at 1pm. Now I have to change the alarm time each time. That means, if I want to set a different alarm besides my recurring alarm for work Mo-Fr 7am I have to change this alarm and hopefully not forget to set it again. I think 2 different alarms are enough, more are welcome.

escoand, 2022-02-02:
Duplicate of #366 and #593 but this was closed without actual outcome for multiple alarms. I stumbled over screenshots of the Apple Watch which looks quite nice: https://tock-cdn.watchaware.com/wp-content/uploads/2016/01/File_002.png

medeyko, 2022-02-05:
I looked at the other discussions. I suppose that the designs proposed there is too complicated, there's a way to make it much easier.

Keep the current alarm screen, but scroll several alarms (probably 3 is a good number) by left-right swipes.

Also there's should be a digit somewhere on the screen to indicate which number this alert is.

bokf, 2022-02-05:
Thanks for your input! In fact I did not search the closed issues, when looking for similar feature requests. As long as the function "multiple alarms" is not implemented the corresponding feature request cannot be closed. That is an Oxymoron https://en.wikipedia.org/wiki/Oxymoron
You can argue that it is too complicated for now, or resources are missing but you cannot close the feature request as it is still valid. A bug you can close, but a not implemented feature request can't be closed, it can only be rated down in priority.

Riksu9000, 2022-02-06:
Feature requests can also be closed if we're not interested in adding said feature. This however would be a good feature to have.

I would prefer having an overview screen with all the alarms in a list, rather than multiples of the alarm setting screens side by side. The design wasn't an issue, but multiple alarms for an initial implementation is a lot to ask for.

medeyko, 2022-02-06:
@Riksu9000
> I would prefer having an overview screen with all the alarms in a list, rather than multiples of the alarm setting screens side by side. 

I see two problems with this:
1. Overview screen requires additional gestures for cases when you need only one alarm. Most times one needs only one alarm, so for most users the additional screen will only mean some unnecessary actions to do, which may annoy. Second and third alarms are rare uses, so they should be implemented in a way that doesn't harm the main one.
2. Another overview screen will take more resources, that are quite limited, as JF002 recently said. Also it needs additional programming, evidently. I think there might be more useful things to put hardware resources and efforts in.

Riksu9000, 2022-02-07:
Without an overview screen you can't get an overview of the alarms, making navigation a bit cumbersome. There would need to be some indication that there are screens to the side, when the screen is already quite full. The way we use the display we can't have a real sideways transition animation, which can make the transition confusing. I don't think an extra click is an issue for setting an alarm. You're probably not in a rush to do it, and even if you were, the overview will help in choosing which alarm to override.

I believe it's fair to use more memory to create a better user experience.

medeyko, 2022-02-07:
> Without an overview screen you can't get an overview of the alarms, making navigation a bit cumbersome.

I strongly disagree: swipes are much more convenient on the small screens than clicking on the alarm in the overview screen.

I don't see problems with indication: small arrows or '<' and '>' would be enough to give a hint that one can swipe.

Moreover I think there should be universal rule for swipes:
Up swipe: return to previous level (exit app).
Left/Right swipe: additional screens on the same level (additional timers/alarms/etc, additonal functions in calculator, previous/next month in calendar, previous/next song in music app).
Down swipe: next level (week overview for calenar, etc.) or mode switch (music app).

> I don't think an extra click is an issue for setting an alarm.

Sorry, but I strongly disagee here as well. For me, this extra click will be really annoying. So annoying that despite I sometimes need additional alarms, I'd prefer to have no additinal alarms than to make extra click to use the main alarm...

> to create a better user experience

So I believe that imposing more unnecessary clicks on users is NOT a better user experience. The most used case should require minimal number of gestures. Any additional functionality should not harm the user experience for the most used cases.

Also, I think that the alarms should be displayed on the face watch (alarms times for the digital one, and color points for the analog one). It's quite useful information that I sometimes miss.

medeyko, 2022-02-07:
@Riksu9000
I think that most people know what alarms they have. As far as I understand, and as far as I read comments in this repository, the most used scenarios are the following:
1. Only one alarm at all (for ad hoc alerts only).
2. Two alarms, one for everyday wake-up (the one that is set up once and for long), and another is for ah hoc alerts.
Note, that even in the second scenario, only one alarm is actively edited.
All other scenarios are quite rare, and so I believe they are not worth handling in case of disturbing the two main scenarios.

So mostly people always know, which alarms are set, unless one uses several "todo" alarms with detailed textual descriptions (but it is not the case with PineTime).

Also I repeat my proposal to indicate the alarms at the watch face. So a user will see the list of alarms w/o even clicking anything!

> You took my comment out of context.

I'm sorry, I didn't intend to do this.

neodc, 2022-04-27:
Hello,

I'm a simple user, but the missing possibility to set multiple alarm is one of the only thing that is making me keep using my old pebble time.

There I use all **10** alarm program available on a weekly basis, I generally need multiple alarm to wake up and keeping track of time in the morning (this already account for 5 of the 10) and switch between different time since I'm partly working from home.

I know that my use case is probably not the most frequent, but could we at least stop telling what the "most used scenarios" is without any research to be based on?

I never saw a smart watch (or any smart thing that can be used as an alarm) without the option to set multiple alarm, that's not random, people use those.

Eloitor, 2022-08-18:
I think it could also be useful to be able to setup an alarm that repeats in 8h or 12h intervals, for medicines.

medeyko, 2022-08-20:
> There I use all **10** alarm program available on a weekly basis, I generally need multiple alarm to wake up and keeping track of time in the morning (this already account for 5 of the 10) and switch between different time since I'm partly working from home.

@neodc - could you please analyse my proposal above and write what do you think? Do you really need an overview screen for your case, or swipes left and right would be enough minimalistic handling of multiple alarms?

neodc, 2022-08-22:
I think an overview screen would be mandatory to quickly enable/disable alarm on a day per day basis. Also handling more that 3 alarm with only swipes look to me like a nightmare to know which alarm the current one is.

timonoj, 2023-07-07:
Was there ever any progress to this? I find the Mi Band 6 implementation to be quite effective. You have a quick overview of all your alarms. Each alarm has a radio button to signal whether you enable it or disable it when tapping it. On the top, an "Edit" button, if you tap it, the overview now has ">" symbols to the right of each alarm, and it will take you to the settings of each individual alarm when tapped. On the bottom, a "+" sign, so you can add as many alarms more as you want. It just works, and it works well. 
[image]

Oh! And they're super easy to sync with Gadgetbridge, so you can create as many as you want there, and they sync in a second. 
Honestly, the ability to easily create/sync both alarms and reminders for random stuff on my phone, that immediately sync to my wristband, this is the only feature I'm currently missing to consider my pinetime "feature complete" enough for me to use as daily driver. I don't care for all the games, watchfaces and other demos people keep adding, but this would be a huge usability quality of life improvement here.

Boteium, 2023-07-07:
@timonoj While UI had been heavily discussed here for multiple times, I don't think it is the most challenging part of implementing multiple alarms. For example, the number of timer in the alarm controller need to be increased. System task need to handle these extra timers. Display app need to handle multiple alarms that goes off at the same time. It requires someone familiar with Infinitime codebase to do it. In contrast, games are much simpler because games may not even need to interact with any system component at all.

Perhaps we can implement alarm preset first, before multiple alarm becomes available. Presets will be much simpler to implement.

timonoj, 2023-07-08:
Thanks for your very fast reply! Sorry, I'm not sure I follow...What would the alarm preset entail?

kvaellning, 2023-09-15:
I just stumbled upon this, since I am planing on getting a PineTime, though I need multiple alarms.

After taking a look in the code, I would pitch the following idea:
- alarms are handled in a structure with an
  - `uint8_t version`
  - `uint8_t id`
  - `bool active`
  - `uint8_t hours`
  - `uint8_t minutes`
  .
  the format should be serializable and allows to be expanded in future (#1333 )
- alarms are stacked upon others:
  - only needs one timer
  - if a timer triggers and other alarms are available, the alarm controller automatically sets up the next timer.
- possible additions:
  - allow setting of alarms at specific weekdays --> encode weekdays by a bitmask

Problem here is that I have done something similiar like this already, but I have no idea how UI design works.

spaetz, 2024-01-05:
One possible solution could be to only show an overview page *if users have set up multiple alarms*. This way the setup remains simple for the majority of users with one alarm and remains convenient for those with multiple alarms.

A sideways swipe could be used to set up a new alarm.

codeguylevi, 2025-02-16:
I think it is a great idea to have multiple alarms!

I use a PineTime, can it work on that?