InfiniTimeOrg/InfiniTime #952: When on a call, rejecting an incoming call ends the current call

pbryan, 2022-01-22:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

While on a call, received an incoming call, and notification; rejecting the call on PineTime ended the current call.

### What should happen instead?

Rejecting the incoming call in PineTime should reject the incoming call.

### Reproduction steps

1. Install GadgetBridge on Android phone.
2. Receive a call on Android phone.
3. While on call, receive another incoming call.
4. Reject the second incoming call in InfiniTime.
5. Observe that the current call has been terminated.

### Version

1.8.0

### Companion app

GadgetBridge 0.64.0

JF002, 2022-01-23:
I think this is an issue for Gadgetbridge. InfiniTime does not know if the call notification corresponds to a "first call", or to a second call coming on top of the 1st one, it just receives a call notification from Gadgetbridge and notifies Gadgetbridge when the "reject" button is tapped.

pbryan, 2022-01-23:
OK, thanks, I'll raise it in GadgetBridge.

closed 2022-01-23 by pbryan

JF002, 2022-01-23:
It would be nice to know if this feature is supported by other devices and how it's implemented. Maybe we'll need to extend the current API to support multiple calls.

pbryan, 2022-01-23:
Here is the [Gadgetbridge issue](https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2568).

pbryan, 2022-01-23:
Will update this issue as I learn more.