InfiniTimeOrg/InfiniTime #953: Drop down list: scrollbar goes over the edge

PhilDevProg, 2022-01-22:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

The scrollbar goes over the edge in the drop down list of the metronome app 

### What should happen instead?

The scrollbar shouldn't go over the edge.

### Reproduction steps

I open the metronom app.
I select press the button with "4 bpb".
I scroll down the list to the bottom.
The scrollbar goes over the edge here.
[image]

### Version

1.8.0

JF002, 2022-03-13:
I guess this bug is related to LVGL (the UI lib used in InfiniTime) and not directly to InfiniTime. 

Unless someone is willing to write a patch for LVGL, this bug might be fixed in a future/more recent version of LVGL, but I don't know when we'll update LVGL in InfiniTime.

Riksu9000, 2022-03-13:
This issue is at least partially caused by incorrect theme configuration in InfiniTime.

JF002, 2022-03-13:
> This issue is at least partially caused by incorrect theme configuration in InfiniTime.

Oh really? I'll re-open it, then!

PR #1102 (2022-05-08): Fix dropdown graphical issues.
The scrollbar would go out of bounds if DROPDOWN_PART_LIST had uneven padding.
~~Enable clip_corner feature to stop the selected item from overflowing.~~ Disabled because it is slow.
`style_pad` wasn't doing anything useful, so I removed it.

[image]

This is the issue that depends on clip_corner to be fixed. We'll just have to live with it.
[image]

Fixes #953
closed 2022-05-08 by JF002