InfiniTimeOrg/InfiniTime #955: Replace the Mono font with something a little nicer

jmaris, 2022-01-24:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

PinetimeLite has a Nice Sans Serif font, could we do the same?

JF002, 2022-01-25:
Do you have a link to such font (preferably one with a license that allow us to use it in an open source project) ?

ainola, 2022-01-26:
There are many such compatible fonts! :)

Perhaps this list would help: https://wiki.archlinux.org/title/Fonts#Latin_script

You can't go wrong with the  Liberation family. Cantarell is used by gnome. Noto is a fan favorite. Lato is a personal one.

geekbozu, 2022-01-26:
This would be a good test for the filesystem as well. I wonder how painful loading a font from the FS would be. If its not exceptionally slow it might be worth writing it to external flash in the next update so we can strip it from the rom in the future!

Riksu9000, 2022-01-29:
I did some testing. I found that the optimal thickness for a non antialiased font is 2 pixels. The current font is about 2-3 pixels and a font with a thickness of a single pixel is too thin to use in some cases. Ideally we would have multiple weights, or at least a regular and a monospaced font. Unfortunately, unless I made a mistake, it seems that a non monospaced font uses ~3700 more bytes due to kerning I suppose.

This is Roboto Medium. These screenshots are a bit blurry, but in reality this looks very nice and sharp. Here we can see another issue however. The font conversion isn't pixel perfect. If we look at the word InfiniTime, we can see that there is a larger gap between i and T than T and i. There are many more cases like this where the spacing isn't as expected.

We need to find a font that is the optimal thickness, preferrably with multiple weights, and looks good at very small sizes. I don't know if the spacing issue is due to the font or the LVGL converter, but it should be taken into consideration.

[image]
[image]
[image]

ainola, 2022-01-29:
Size was something I was a little concerned about. These look super good though! Much more readable.

JF002, 2022-01-31:
Yes, I think it looks good, but as you say, the spacing looks inconsistent. The word "format" on the 2nd screenshot look like "forma t" with a space before the 't'.

medeyko, 2022-02-07:
I think that Jetbrains mono font is more stylish. I'd prefer it not to be removed.

Also variable width font should not be used on watch faces and in most apps, to ensure that the position of characters is constant. It is especially important for dynamically changing texts.

Also I believe that variable-length font inevitable requires more hardware resources, both memory and CPU.

ainola, 2022-02-07:
On 2022-02-06 22:05, medeyko wrote:
>Also I believe that variable-length font inevitable requires more hardware resources, both memory and CPU.

I'd be interested in getting some benchmarks because this makes me think 
of the "SSL takes more hardware resources" that may have been valid on 
older hardware but perhaps is not the case nowadays.

But that is true that it would cause more maintenance to ensure proper 
kerning, etc.

jarinox, 2023-06-27:
> Yes, I think it looks good, but as you say, the spacing looks inconsistent. The word "format" on the 2nd screenshot look like "forma t" with a space before the 't'.

Even with this inconsistency I think the proposed font looks much nicer than the current monospace font.