InfiniTimeOrg/InfiniTime #957: Allow multiple presses and holds for more varied user interactions

Soundtoxin, 2022-01-25:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Allow holds, double taps, etc. to make more use of the button

### Description

I thought of this after playing with flashlights lately. Most enthusiast flashlights have one button, but manage to cram a ton of functionality into it. Pressing it once while off and once while on does something different, holding the button does something different, there are double and triple taps and sometimes more. There's even the fantastic [Anduril firmware](https://ivanthinking.net/thoughts/anduril2-manual/) where you can do things like press the button 7 times to configure auxiliary LEDs, and they have a system of clicks and holds where "3C" means 3 clicks, but "3H" means two clicks followed by a click and hold. This is enough to have a sort of "menu" system where you can click a certain number of times to get to configuring a specific thing, and then once there, a simple tap or hold can change a setting.

A more simple comparison would be the media controls on earbuds with in-line remotes. Commonly you can play/pause with a single press, skip forward a song with a double press, and skip backward with a triple press. Touchscreens are very limiting when it comes to accuracy and use without looking directly at the thing, so making more use of the button could really improve some things I think.

Riksu9000, 2022-02-21:
InfiniTime can already use click, long press, longer press and double clicks to perform actions. Unllike flashlights, we have a touch screen where we can display buttons and text for easier use. What kinds of actions do you think should be mapped to the button?

Soundtoxin, 2022-02-21:
On my Pebble with the 4 buttons I had shortcuts to most-used apps on the long-press of things, like holding the okay button to get to the stopwatch and so on. Very usable without even looking at it. I think with just one button it would be hard to recreate that, but you could at least make it so a long press acts as "okay" and then double/triple could be up/down for navigating menus maybe.

Riksu9000, 2022-02-21:
Navigating the UI with button presses is not really practical when you can just swipe on the screen and it's much faster. There's been some talk about having shortcuts to apps, but really it only takes a second to open any app with just touch gestures.

Soundtoxin, 2022-02-21:
Touchscreens lack accuracy and tactile feedback. Buttons let you use things with your eyes closed, and assuming the UI doesn't change you can remember you have to go up 2 times on one screen, press okay, down, okay again, etc. and you can blindly get to commonly used stuff that way.

If it were up to me, the PineTime would just have more buttons. It's really sad that it doesn't. I guess I'm not sure if it's something worth trying to fix in software or not.

Riksu9000, 2022-02-21:
Sure, but we only have a single button. Navigating a complex graphical interface solely with a single button doesn't make sense. The button can be used in conjunction with the touchscreen while navigating, as it generally acts as a back button.

Riksu9000, 2022-02-26:
Do you have other suggestions, or has this issue been resolved?

Riksu9000, 2022-02-26:
I don't really understand what exactly it is that you want since you're mostly comparing to other things and these ideas are not directly applicable on the PineTime and the InfiniTime interface, so I don't know how to solve this. I see that you wish for something more like the Pebble, but PineTime and InfiniTime are fundamentally different than that and I don't think this single button navigation would serve you like you might expect it to.

Let's say you wanted to open the timer, you'd have to doubleclick to move to the app screen, and then double click 5 more times to move a cursor to the timer app and then hold the button to open that app. It's also not just pressing the button 12 times. You'd have to wait between each double click. Meanwhile you can also do this in two touch inputs.

I read the original post as generally wishing for making more use of the button, which I'm not opposed to, but such an open ended issue isn't easy to solve, since it's not clear what should be done for this to be considered solved. Closing this doesn't mean that nothing will ever be done about this. Someone can still open a new issue to propose new ideas for consideration, but for now if we can't figure out what should be done about this, I think this should be closed.

JF002, 2022-02-28:
@Soundtoxin I agree with @Riksu9000 and I don't really understand how you would like this new feature to be integrated in InfiniTime...
You suggest we could use the button the same as on flashlights and Pebble devices. I think we are OK with that but... how do you translate this idea in InfiniTime? Do you have any specific idea or use-case in mind? We cannot do much if we don't even know what to do...

Soundtoxin, 2022-02-28:
I believe I already mentioned using multi-press for up/down and okay for a touch-less blind-friendly setup. No need to drag this out. My idea was rejected. I won't close the issue, but I thought one of you would've by now.

JF002, 2022-02-28:
Your idea is not rejected, but we cannot understand what you have in mind. It's impossible for us to understand what code we could write to implement your idea... it's far too vague.

Up/down : where exactly? In an app, a menu, the watchface ? For a specific action or to replace the up/down gesture on the touchpanel? Which combination of short and long presses would you like? (knowing that 1 short, 2 shorts, 1 long press and 1 very long press are already used in InfiniTime)
"Okay": you mean for an "OK" button?

Soundtoxin, 2022-02-28:
Interesting. I didn't think it was particularly unclear. Yes, buttons, all buttons, and I have navigation of menus in mind, but also within apps. Allow someone to wake their watch, get to the stopwatch app (just an example), and start/stop/lap on that stopwatch with various presses of the button.

1 short remains back to stay compatible with current usage. I hadn't seen 2 shorts / one long used that I could recall, I thought they were unused. Yes, an "OK" button, confirmation button, the opposite of back, similar to enter. I just have the four pebble buttons in mind here with what I'm saying, fyi. 

One press already wakes the screen, then if you're on the watchface/homescreen, either back or OK could open the apps menu. from here you use presses to move around the menu and select the app. Inside the app chosen, ideally this sort of navigation would continue to work.

JF002, 2022-02-28:
Here are the current functions of the button:
 - 1 short press
   - When running : back or go to sleep
   - When sleeping : wake up
 - 2 short presses : open notifications
 - 1 long press : go back to the watchface
 - 1 longer press : open the "about" info pages
 - 1 even longer press : reset

Now, we'll need more precise info than "various presses" for "various actions".
Here's the level of details we would need:
`While on the watch face 3 shorts and then 1 mid long presses to open the Stopwatch app`
`In the Music app, 2 shorts and 1 long to increase the volume`
`Anywhere, 2 longs and 1 short presses to go down`

Riksu9000, 2022-02-28:
I mean I'm still not on board with this idea since it's very complicated and I don't see any benefits. There doesn't need to be multiple ways to do every single thing on the watch. There is no reason to have confusing alternative navigation methods so one can refrain from using the touch screen on a touch screen device. I think this should be considered out of scope. I'm not sure continuing this discussion is going to lead anywhere useful.

JF002, 2022-03-02:
I agree that the examples I wrote make very little sense and do not look practical at all, and I hoped the OP would provide better example from his idea but they didn't answer. I'm closing this issue. Feel free to re-open with relevant information.

closed 2022-03-02 by JF002