InfiniTimeOrg/InfiniTime #970: Nothing changes after building and uploading output OTA?

SomethingGeneric, 2022-01-29:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Firmware version is the same

### What should happen instead?

I'd expect to see the version number I overwrote for testing

### Reproduction steps

I wrote this script to automate stuff like SDK and toolchain path. Seems to build fine. Then, it takes the bin image and makes a DFU zip for the firmware (as I understand it). 
```sh
#!/bin/bash

# In addition to toolchain and sdk as below, make sure to:
# pip3 install --user adafruit-nrfutil

TC_PATH="/home/matt/gcc-arm-none-eabi-9-2020-q2-update"
SDK_PATH="/home/matt/Downloads/nRF5_SDK_15.3.0_59ac345"

if [[ -d build ]]; then
    rm -rf build
fi

mkdir build
cd build

# We're building DFU straight-away since my dumbass doesn't have a programmer thingie.
# See https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/buildAndProgram.md
# if you're not fucking braindead like me, and want to just build the binaries not a full
# OTA package.
cmake -DARM_NONE_EABI_TOOLCHAIN_PATH=${TC_PATH} -DNRF5_SDK_PATH=${SDK_PATH} -DBUILD_DFU=1 ../ 

make -j pinetime-app

adafruit-nrfutil dfu genpkg --dev-type 0x0052 --application src/pinetime-app-*.bin dfu.zip

printf "File name: "
read NAME

mv dfu.zip ../pinetime-app-${NAME}.zip

cd ../

rm -rf build
```

However, when flashing the output-ted zip with nrfConnect (I don't have any of the debuggers :cry: ), and then a restart, none of my changes show up? 

Since everything seems to have worked correctly (and seeing I purge the `build/` directory each time, I have no idea why I cant see the version I overwrote nor the rename I did to the "Shake Calib." menu item in settings?

### Version

1.8.0

### Companion app

nrfConnect 4.24.3 on Android 12 on S21+ 5G

SomethingGeneric, 2022-01-29:
Sorry all turns out I misread the two different markdown files and just needed to use `make pinetime-mcuboot-app`. :upside_down_face: 

Y'all might get a PR with a workflow suggestion for complete knobheads like myself testing code via OTA and praying.

closed 2022-01-29 by SomethingGeneric

JF002, 2022-01-31:
Yes, you are right, you need to use target `pinetime-mcuboot-app` to build the firmware that'll work with the bootloader. The bootloader needs the firmware to be linked with a start address at 0x8000 (which is done with `make pinetime-mcuboot-app`. The target `pinetime-app` builds the firmware link at address 0x00. This one runs without the bootloader, but can only be flashed with an SWD debugger. If you OTA it, the bootloader will just revert it because it's not a "bootable" firmware.

Also, did you know that you can generate the DFU file directly with cmake and make? When you run cmake with  `-DBUILD_DFU=1` and then `make pinetime-mcuboot-app`, you'll find the DFU file generated in the `src` folder of your build directory :

```
    ~/git/PineTime/build   @688dc2fe *7 !1 ?8  cmake -DARM_NONE_EABI_TOOLCHAIN_PATH=/home/jf/nrf52/gcc-arm-none-eabi-9-2020-q2-update -DNRF5_SDK_PATH=/home/jf/nrf52/nRF5_SDK_15.3.0_59ac345 -DBUILD_DFU=1  ..                INT ✘ 
-- The C compiler identification is GNU 11.1.0
-- The CXX compiler identification is GNU 11.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
PROJECT_GIT_COMMIT_HASH_SUCCESS? 0
BUILD CONFIGURATION
-------------------
    * Version : 1.8.0
    * Toolchain : /home/jf/nrf52/gcc-arm-none-eabi-9-2020-q2-update
    * GitRef(S) : 688dc2fe
    * NRF52 SDK : /home/jf/nrf52/nRF5_SDK_15.3.0_59ac345
    * Target device : PineTime
    * Debug pins : Disabled
    * Build DFU (using adafruit-nrfutil) : Enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jf/git/PineTime/build
    ~/git/PineTime/build   @688dc2fe *7 !1 ?8  make -j pinetime-mcuboot-app                                                                                                                                                       ✔ 
[...]
[100%] Linking CXX executable pinetime-mcuboot-app-1.8.0.out
Memory region         Used Size  Region Size  %age Used
           FLASH:      397836 B       480 KB     80.94%
             RAM:       54848 B        64 KB     83.69%
post build steps for pinetime-mcuboot-app-1.8.0
   text    data     bss     dec     hex filename
 397836     980   53860  452676   6e844 pinetime-mcuboot-app-1.8.0.out
post build (DFU) steps for pinetime-mcuboot-app-1.8.0
Zip created at pinetime-mcuboot-app-dfu-1.8.0.zip
[100%] Built target pinetime-mcuboot-app
    ~/git/PineTime/build   @688dc2fe *7 !1 ?8  ls src                                                                                                                                                                      ✔  9s  
CMakeFiles  Version.h            libQCBOR.a     liblvgl.a    libnrf-sdk.a                    pinetime-mcuboot-app-1.8.0.hex  pinetime-mcuboot-app-1.8.0.out      pinetime-mcuboot-app-image-1.8.0.bin
Makefile    cmake_install.cmake  liblittlefs.a  libnimble.a  pinetime-mcuboot-app-1.8.0.bin  pinetime-mcuboot-app-1.8.0.map  pinetime-mcuboot-app-dfu-1.8.0.zip
```

SomethingGeneric, 2022-01-31:
Yeah, that's what I ended up doing to avoid needing another step :)