InfiniTimeOrg/InfiniTime #972: More readable notification for "old eyes"

tomechio, 2022-01-29:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

black/white notifications or forced upper case letters

### Description

Hi, 

As a middle age person my sight isn't as sharp as it used to be when I was a kid. 
The notifications are quite difficult to read unless I strech my arm as far as I can. 

So it would be nice if:
- the text could be forced to upper case letters
or:
- user could select white text on black background instead of yellow text in gray background.

[image]

arrrlaska, 2022-01-31:
Seconded. The notifications cause a lot of eyestrain without reading glasses. 
A black background would totally fix this.

Contrast is needed. Please, please tweak this, devs.

arrrlaska, 2022-01-31:
Semi-related: https://github.com/InfiniTimeOrg/InfiniTime/issues/647

SomethingGeneric, 2022-01-31:
Hey guys I'm not much of a C/C++ dev, but I can offer that my fork (made mostly because I prefer navy to green for accent color), also includes white notification text, and when I have some time later, I'll also look into text size. If I can figure out text size I can also make a PR against the main version, but tbh most of the stuff in my fork is personal preferences as I said.

SomethingGeneric, 2022-01-31:
im also happy to provide a dfu zip, as that's how i've been testing myself

tomechio, 2022-02-02:
> im also happy to provide a dfu zip, as that's how i've been testing myself

Thank you for your suggestion. I just managed finally to get the docker to work by updating cbor to cbor2 in boot_record.py
So to learn little bit C programming I first try to modify notifications by myself.

arrrlaska, 2022-02-08:
> im also happy to provide a dfu zip, as that's how i've been testing myself

I'd LOVE a zip, if you've tested it (I only have a sealed unit) and it's ok. :smiley: 

I also think it's a basic usability issue they should really address.

tomechio, 2022-02-09:
You can try this:
https://github.com/tomechio/InfiniTime/releases/tag/1.8.0

It's running on my sealed device.

arrrlaska, 2022-02-09:
@tomechio wow, that is SO much more readable!!

I wish you'd do a pull request just for a darker background color (minimal change to code)

Thank you so much! 🤓

tomechio, 2022-03-06:
I have now played around little bit more with the notifications and seems that only changing font color from yellow to white makes the notifications more readable.

ajack2001my, 2022-04-08:
Can I put my request here instead of a new request?

Anyways, If possible to whoever is looking after this request, to add the ability to clear individual notifications or all notifications at once.

Sec42, 2022-04-18:
I played a bit with the code & made two changes: 
a) make the notification more readable by using white color & a new slightly larger font
  https://github.com/Sec42/InfiniTime/commit/221a6a3e35c17c6876066b345446720bcc0b5e14
b) ability to delete the newest notification by swiping left on it
  https://github.com/Sec42/InfiniTime/commit/761bd2222406023d07efd1fd9a97c0d87e3f82f6

Code is in this branch https://github.com/Sec42/InfiniTime/tree/notification-improvements

It could probably be improved, but for now I'll run it on my watch for a while to see if I'm happy with it.

arrrlaska, 2022-04-19:
> I played a bit with the code & made two changes: ...

That's super! Do you have a build available?

Sec42, 2022-04-20:
I have attached the build as a release here: https://github.com/Sec42/InfiniTime/releases/tag/1.9.0

arrrlaska, 2022-04-20:
> I have attached the build as a release here:

That is WONDERFUL. thanks so much! Love the larger font, too!

arrrlaska, 2022-07-19:
Does anyone have a readability build of 1.10.0?
@Sec42 ? 😄

Sec42, 2022-07-28:
There were some (incompatible) changes in 1.10. I have re-implemented the larger notifications & published them here:
https://github.com/Sec42/InfiniTime/releases/tag/1.10.0

arrrlaska, 2022-07-28:
> There were some (incompatible) changes in 1.10. I have re-implemented the larger notifications & published them here: https://github.com/Sec42/InfiniTime/releases/tag/1.10.0

Love it! Thank you so much! :smile:

PR #1252 (2022-08-08): Notification swap text colors
Improves visibility. Also reduced code duplication slightly. Fixes #972.

[image]
closed 2022-08-08 by Riksu9000