InfiniTimeOrg/InfiniTime #981: Duplicate notifications from SMS messages with Signal set as unified SMS messenger

calhix, 2022-02-01:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

I get two notifications of the same thing when receiving an SMS through Signal

### What should happen instead?

Only see one notification

### Reproduction steps

Set Signal as default SMS app
Receive an SMS from somebody

### More details?

I brought this up in #934 and thought I'd make a separate post.

### Version

1.8.0

### Companion app

Gadgetbridge v0.65.0

Avamander, 2022-02-01:
https://codeberg.org/Freeyourgadget/Gadgetbridge/wiki/Notifications

closed 2022-02-01 by Avamander

infinitytec, 2023-03-06:
FYI for anyone having this issue and searching through reports:
I had this issue with GadgetBridge as well. Turned out the solution was to disable the SMS notification in GadgetBridge. My text app would then be pushing a notification as well, so that's why I would get two. I also noticed this with phone calls.

You could use the built-in GadgetBridge SMS notification but you would need to then disable the notifications from your SMS app.