InfiniTimeOrg/InfiniTime #984: Failure to operate the alarm clock on charging.

13werwolf13, 2022-02-03:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

When the watch is on charge, there is no point in triggering the alarm clock. 

### Description

Last night before going to bed, I discovered that the battery in my PineTime was almost completely depleted. 
I put the watch on charge and went to bed. In the morning Waking up of course I did not run immediately after the clock, first I drank coffee, brushed my teeth, read the mail and the news. 
All this time, the clock brought my parrots to a state of panic as it thundered a vibrating motor on the metal body of the server without stopping. 
This leads me to the idea that it is worth adding functionality that will turn off the alarm clock when the watch is on charge and turn on again when the watch is off charging. 
P.S.: Sorry for my English.

SteveAmor, 2022-02-03:
Is this covered by #446 ?

13werwolf13, 2022-02-03:
> Is this covered by #446 ?

Partially yes. But turning off the vibration will not turn off the alarm clock, just make it soundless.

Riksu9000, 2022-02-14:
I don't think it's a good idea to disable the alarm while the watch is charging. When the alarm is set, it is expected to ring when it is time, no matter what. I also don't think it is worth having a setting for this. If the user isn't going to be around for the alarm, then the alarm should be disabled. A change was recently made to make the alarm stop after ringing for 60s in case the user isn't around to stop it.

I do however think it may be worth lowering the vibration strength while charging, to reduce the awful noise it makes. This is more related to #446.

closed 2022-06-12 by Riksu9000