InfiniTimeOrg/InfiniTime #985: Multiple timers

parcelcat, 2022-02-03:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Support adding, deleting, starting, and stopping multiple timers. Allow adding, deleting, starting, and stopping other timers while existing ones are running.

### Description

Having multiple alarms would be a useful addition to InfiniTime. Someone who is multitasking could set different timers for different tasks and use those timers simultaneously.

Example implementations:
- [Apple watchOS](https://support.apple.com/guide/watch/timers-apdf448955b2/watchos): Apple Watches have a touchscreen like the PineTime. The PineTime has a physical button instead of the Apple Watches' digital crown.
- [Multi Timer](https://github.com/smallstoneapps/multi-timer) for Pebble: Pebble watches have no touchscreen, unlike the PineTime. Pebble watches have three physical buttons compared to the PineTime's one physical button.

This feature request is similar to #950 (multiple alarms).

medeyko, 2022-02-05:
I'd repeat my proposal for alerts. I think it would equally suit timers.

Keep the current timer screen, but scroll several timers (probably 3 is a good number) by left-right swipes.

Also there's should be a digit somewhere on the screen to indicate which number this timer is.

minacode, 2022-02-06:
Or make a scrollable list like the settings menu. And tapping an entry brings you to the timer screen that we have now.

Riksu9000, 2022-02-06:
It makes sense to have multiple alarms, but timers? Seems to me like it would be very overwhelming having to do so many things at the same time that multiple timers would be required.

The timers would need to be labeled somehow to be able to differentiate between them when they end.

I think I would also prefer having an overview screen rather than multiples of the timer setting screen side by side, if this were to be implemented.

The samsung galaxy gear link is about alarms, not timers btw.

medeyko, 2022-02-06:
> I think I would also prefer having an overview screen rather than multiples of the timer setting screen side by side, if this were to be implemented.

Like with the alarms, I see two problems with this too:
1. Overview screen requires additional gestures for cases when you need only one timer. Most times one needs only one timer, so for most users the additional screen will only mean some unnecessary actions to do, which may annoy. Second and third tirmers are rare uses, so they should be implemented in a way that doesn't harm the main one.
2. Another overview screen will take more resources, that are quite limited, as JF002 recently said. Also it needs additional programming, evidently. I think there might be more useful things to put hardware resources and efforts in.

parcelcat, 2022-02-07:
> It makes sense to have multiple alarms, but timers? Seems to me like it would be very overwhelming having to do so many things at the same time that multiple timers would be required.

An example of when multiple timers would be useful is when the user is cooking multiple courses for a large meal, and needs to keep track of the cooking or preparation time for each dish separately.

> The samsung galaxy gear link is about alarms, not timers btw.

Thank you, I've replaced it with a link to a third-party app for the Galaxy Gear called MultiTimer Pro. This app uses icons and colors to help the user distinguish between timers.

kyeastmood, 2023-12-28:
Multiple timers are really useful, I can have something boiling in the kitchen while brewing my tea or exercising in the mean time.

codeguylevi, 2025-02-16:
That would be helpful.