InfiniTimeOrg/InfiniTime #986: BLE is not enabled unless explicitly rebooting the watch

bergercookie, 2022-02-06:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Tried connecting the PineTime watch with any of the companion apps to e.g. synchronize the time and it wasn't getting detected.

### What should happen instead?

get detected over bluetooth from the get-go

### Reproduction steps

* Get a sealed PineTime watch. Mine came with the 1.6.0 release
* Power it up for the first time
* Charge it for a short while (mine came dry out of the box)
* Try synchronising the time with any of the WebBleWatch app,  Siglo or Gadgetbridge. None should be able to detect the watch over BLE

In order for the BLE detection to workI had to explicitly restart the watch by long-pressing its physical button.
After that BLE detection worked as expected.

### More details?

Thanks for your work. This is a great product.

### Version

v1.6.0

### Companion app

You should repro this with any of the companion apps.

Riksu9000, 2022-02-06:
There have been improvements to bluetooth connectivity in newer firmware releases. Please upgrade the firmware on your watch to the latest version (1.8.0 at the time of writing).

closed 2022-02-06 by Riksu9000

bergercookie, 2022-02-06:
Sounds good. As I said, I solved my issue by restarting the watch manually.
I think it would be helpful adding such note in the relevant documentation

JF002, 2022-02-06:
In theory, it shouldn't be needed to restart the watch to enable BLE. But, as @Riksu9000, we fixed a few issues regarding BLE connectivity in newer versions, so those bugs are hopefully fixed now :).

By the way, we are working on a complete rework of the documentation here : #838