InfiniTimeOrg/InfiniTime #988: Zero character

medeyko, 2022-02-07:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

The dot inside the zero character is glued to the 0 itself. That resembles 8 and makes reading more difficult.

### What should happen instead?

There dot inside 0 should not glue to the boundary.

### Reproduction steps

See for example 2022 on the digital watch face.

### More details?

Here's a sample image:

[image]

Zero in 20 is current 0 character. Zero standing alone is how I propose to change the 0 character.

To make this, the jetbrains_mono_bold_20.c should be edited:

    /* U+0030 "0" */
    0x3f, 0x1f, 0xef, 0x3f, 0x87, 0xe1, 0xf8, 0x7f,
    0xdf, 0xf7, 0xe1, 0xf8, 0x7e, 0x1f, 0xcf, 0x7f,
    0x8f, 0xc0,

should be replaced with
    /* U+0030 "0" */
    0x3f, 0x1f, 0xef, 0x3f, 0x87, 0xe1, 0xf8, 0x7e,
    0xdf, 0xb7, 0xe1, 0xf8, 0x7e, 0x1f, 0xcf, 0x7f,
    0x8f, 0xc0,

(there are two changes: 7f -> 7e and f7 -> b7.)

I understand that the process of the font generation is automated, but I think that this manual change worth doing.

### Version

v1.8.0

escoand, 2022-02-07:
This is maybe fixed with #955.

medeyko, 2022-02-07:
@escoand 
I don't think that the Jetbrains mono font should be removed. Anyway, #955 issue will be discussed some time, I think. The fix I proposed is quite simple and may be introduced in 1.9.0 already.

Riksu9000, 2022-02-07:
I think the reason this hasn't been fixed is that the fix may easily get lost when someone updates the font. But I think this may be worth fixing anyway, as long as it is well documented and we make sure the fix doesn't get lost with future updates. Since you have the solution to the issue, would you like to create the PR for it @medeyko ? Make sure to update https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/src/displayapp/fonts/README.md

medeyko, 2022-02-07:
@Riksu9000 
Thank you! I made #989 PR, hopefully it's fine...

JF002, 2022-03-13:
#989 is merged, I guess I can close this issue.

closed 2022-03-13 by JF002

medeyko, 2022-04-14:
Great! Thank you very much!

@Riksu9000: sorry for not replying earlier, I reread your comments, and it seems I made everything according to them as far as I understand. If I could do more, please write.