InfiniTimeOrg/InfiniTime #999: metronome modifications

josiahb, 2022-02-16:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Two small changes I'd love to see with the metronome

### Description

1. extend the lower end of tempos accessible via the slider to 20. Though I'm not certain, I think this may simply involve changing 

`lv_arc_set_range(bpmArc, 40, 220);`

to

`lv_arc_set_range(bpmArc, 20, 220);`

2. a 'zero' beats per measure option, so that there are no strong pulses.

Riksu9000, 2022-02-17:
What's your use case that requires extending the range? 0bpb option isn't necessary. The 1bpb option should just do medium vibrations instead.

josiahb, 2022-02-17:
My use case for an extended tempo range: I'm a musician and set slow tempos fairly often to train my inner pulse (i.e., having pulses at the beginning of every bar, or every two bars). 20bpm is about as slow as I've found myself able to effectively use. A more nuanced metronome might have an option for only pulsing on downbeats, but I don't expect that kind of granularity in this case. A lower edge to the slider should be sufficient. 

As for bpb: I agree, though it's fairly common for standalone metronomes to have a separate setting for no strong pulses as opposed to strong pulses on each beat. Maybe that's more subtle than necessary here.

Riksu9000, 2022-03-13:
From a quick browsing, it seems like 40bpm is the standard lower limit on metronomes. I think your use case is valid, but it would technically be using the app in an unintended way. If we break the standard and allow the app to be used in unintended ways, how do we know that someone won't come along and say that it now needs to go down to 10bpm, and so on? Only pulsing on downbeats sounds like a decent solution to me, but it will complicate the interface. Maybe the bpb dropdown should instead open a settings screen that has these extra options.

Danimations, 2022-04-01:
I'm also a musician and the current lower limit of 40 bpm works fine for me.

closed 2023-02-22 by Riksu9000