InfiniTimeOrg/InfiniTime #1008: Can't pair with GadgetBridge after uninstalling and reinstalling GadgetBridge

duncan-bayne, 2022-02-23:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

I uninstalled then reinstalled GadgetBridge; now my PineTime can't pair

### What should happen instead?

I'd expect to be able to pair my PineTime

### Reproduction steps

1. Install GadgetBridge.
2. Pair PineTime with GadgetBridge.
3. Uninstall GadgetBridge (_without_ removing PineTime from GadgetBridge first).
4. Install GadgetBridge.
5. Configure required permissions in Android.
6. Wake up PineTime.
7. Click Start Discovery in GadgetBridge.

### More details?

At this point I'd expect the PineTime device to show up in Gadgetbridge; however, it doesn't.

I've also tried restarting both the PineTime and Gadgetbridge.  If I configure GadgetBridge to display all BTLE devices (including incompatible devices) it lists many; so I suspect the problem isn't with BT / BTLE or Android.

Versions of things:

- PineTime 1.8.0
- Android 11 (Google Pixel 2)
- GadgetBridge 0.65.0

### Version

1.8.0

### Companion app

GadgetBridge 0.65.0

Avamander, 2022-02-23:
Unpair the PT from Android settings and/or reboot the PineTime.

duncan-bayne, 2022-02-23:
Ah son of a ... 

Thank you @Avamander :)  It was somewhat hidden.  In case someone else hits this issue, on Android 11:

1. Open Settings.
2. Tap Connected devices.
3. Tap 'See all'.
4. _There_ you'll see the InfiniTime device.
5. Tap the gear icon to the right of the InfiniTime device.
6. Tap Forget.

Then the PineTime will pair as expected.

closed 2022-02-23 by duncan-bayne