InfiniTimeOrg/InfiniTime #1016: Increase the transmit power on watch so that the range is more than 150 feet

tucuongbrt, 2022-02-28:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

use ble_phy_txpwr_set(tx_pwr_lvl)

### Description

Hi all,
I try to connect the Watch to the Nrf Tools App more than 150 feet (currently only about 15 feet in a house). I read some document and thread about Nimble, I add this code:
/* Set tx power level. */
  rc = ble_phy_txpwr_set(tx_pwr_lvl);  // tx_pwr_lvl == 4
  memset(&fields, 0, sizeof(fields));
  fields.tx_pwr_lvl = (uint8_t) tx_pwr_lvl;
in function StartAdvertising() of NimbleController.cpp.
The above doesn't seem to have an effect on the contents of the advertising packet; it still advertises around 15 feet.
The comment in https://github.com/apache/mynewt-nimble/blob/089f0d849a49a2a6c69f1e1f9103174b81af8b1a/nimble/drivers/native/src/ble_phy.c  implies that this should be handled automatically, and the provided value is ignored. Is this implemented as described, or am I doing something wrong?
The actual tx power does change, I see the expected difference in received signal between +4, 0, -12, -20.
Thanks all

trman, 2022-03-03:
hardly usable , since it would drink feeble pinetime battery very fast

Boteium, 2022-12-03:
I was thinking about long range mode (coded phy) but sadly nrf52832 does not support it.
Perhaps future generation of pinetime will upgrade to nrf52840 platform.

jones139, 2024-04-05:
I know this is an old issue, but I am having trouble with the range of the PineTime being too low, giving unreliable data transfer to the phone unless the watch is very close to the phone.

I can see that in NimbleConroller.cpp, startAdvertising(),  we set fields.tx_pwr_lvl; to be automatic.

If I change it to 4 db (which I think is the maximum for the radio), will this set the radio power all the time, or just when it is advertising?   

Looking at the rssi value on nrf connect it does not appear to make much difference, but my app does seem to be doing a bit better.  

I am not used to working with BLE so would appreciate any thoughts on what I can change to improve things (I am happy to sacrifice battery life as long as it will run for at least 12 hours).