InfiniTimeOrg/InfiniTime #1020: Remotely lock/unlock companion device

0x1a8510f2, 2022-03-03:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Ever had someone pick up your phone when you left it unlocked, or straight up snatch it out of your hand? This is a problem no more!

### Description

While I admittedly don't require this feature often, when I do, I really regret not opening this issue earlier.

Essentially, I believe it would be cool to have an app on the watch which is able to send a command to the companion app requesting that it lock or unlock the companion device (not sure if the latter is technically possible on most mobile devices, but it should be on desktop, definitely is on Linux). Given the recent encrypted communication feature, this should be secure, but the companion app will likely offer an option to enable or disable this anyway.

Here are the possible use cases:

- Got a long password or PIN for better security but tired of typing it each time? Just unlock your device with your watch!
- Annoying friends who steal your phone? Lock it from your watch!
- Left your computer or laptop unlocked but you already walked 10 steps away from it and don't want to walk back? Lock it from your watch!
- A friend needs to use your laptop but you don't want to type your password in front of them? Unlock it with your watch! And lock it again if they try to do something they shouldn't do!

This app should be relatively small and easy to implement on the InfiniTime side, so I'm hoping that the usefulness:difficulty ratio is enough for this to be accepted :P. Even so, I'll wait for some comments here before opening issues for companion apps.

Possibly related, is a lock feature for InfiniTime, but it's not really useful unless this is added, so I'll wait before opening that issue also.

Riksu9000, 2022-03-05:
The connection range is very short, so if someone were to seriously steal your device, they'd probably have run out of range before you could do anything. #1016 is about increasing the range. As for unlocking, something like the Google Smart Lock would be nice, which keeps the phone unlocked while in range of a smartwatch. Maybe it's already possible, but I'm not able to test it.

0x1a8510f2, 2022-03-05:
>The connection range is very short

I'm aware. While that is unfortunate, I think it's fine for most of the above use cases. I didn't mean for this feature to prevent theft, only to let the user lock or unlock their device without needing to touch it and one way among others this is useful is to lock your phone when a friend/sibling/parent jokingly takes it, as this situation is what initially gave me the idea.

>As for unlocking, something like the Google Smart Lock would be nice, which keeps the phone unlocked while in range of a smartwatch

Fair point, I didn't even think of smart lock. But I still think this feature is useful because it locks and unlocks the device on-demand which is more secure than just proximity.

huckvrni, 2022-03-12:
Could work well with #1032

0x1a8510f2, 2022-03-12:
>Could work well with #1032

Certainly, that would be awesome!

Arsen6331, 2022-03-19:
I am not sure how well this would work for iOS and Android, but I do know it will be a major problem for Linux. There is no easy and universal way to unlock a session without root privileges, which as far as I know, no Linux companion app has. You could make a daemon that does that for you, but it would massively decrease the security of your system. 

Android does seem to have a way to do this, but it seems like a major vulnerability. From what I can tell, iOS doesn't allow it, and Apple rejects your app if you do it. I might be wrong though, this was from a quick search.

huckvrni, 2022-03-19:
I used to use the automate app on android so at the very least I know it can work there...
https://llamalab.com/automate/doc/block/device_lock.html

Arsen6331, 2022-03-19:
> I used to use the automate app on android so at the very least I know it can work there... https://llamalab.com/automate/doc/block/device_lock.html

Yeah, it does work on Android, but it won't on iOS and Linux. Honestly, I don't know why Android allows this, it creates a very easy attack vector for hackers.

Arsen6331, 2022-03-19:
In fact, putting this in IninfiTime could also present a certain risk. If an attacker can successfully impersonate the watch, then they can easily unlock your phone.

huckvrni, 2022-03-19:
No no you got it all wrong...
You can ONLY lock the phone this way, the only thing you can do with a trusted Bluetooth device is to make the phone stay unlocked after unlocking it.

All that aside, the moment any attacker has physical access to your machine there is really nothing that will stop him, a good password will at most delay him for a bit...

Arsen6331, 2022-03-19:
> No no you got it all wrong... You can ONLY lock the phone this way, the only thing you can do with a trusted Bluetooth device is to make the phone stay unlocked after unlocking it.
>
> All that aside, [...]

Ok, that's better, but it still isn't going to work on Linux or iOS. Maybe you can make iOS stay on and never go to sleep, that would be the only way there. On Linux, this would be entirely impossible.

huckvrni, 2022-03-19:
Yeah I guess in any case it's the companion apps job to support this feature first, we should just enable the api

tlhonmey, 2022-06-08:
There are already a few programs which will perform various actions on Linux and Android when a particular bluetooth device goes in or out of range.  That's probably the best way to handle the snatching case since likely they'll be out of range before the user can react.

There doesn't seem to be anything explicitly for this kind of action in the bluetooth profile specifications, but perhaps the HID profile coupled with some of the extra HID keycodes that are available on multimedia keyboards and suchlike.  Those have buttons for sleep, hibernate, and a number of other things.  Not sure how broad the support is, but an app that could be configured with buttons to send specific key-presses could then have shortcuts set on the attached device to do any number of broadly useful things.

Arsen6331, 2022-06-08:
> There are already a few programs which will perform various actions on Linux and Android when a particular bluetooth device goes in or out of range. That's probably the best way to handle the [...]

Yeah, the issue here is that most OSes will not allow you to simply lock and unlock the device at will. There's no way to do it on iOS and Linux. The only iOS app that does this is Apple's own Watch app, and they don't expose the functionality to other apps. On Linux, some desktop environments have the ability to unlock via an app, but most don't, and the only way that it _might_ be possible in a DE-agnostic way is by using root privileges which is not secure and very much not recommended, and even then, some DEs won't listen for that and the locker won't be killed. That's not even to mention people using standalone lockers like i3lock with a WM rather than a full DE, like I do.

tlhonmey, 2022-06-08:
> Yeah, the issue here is that most OSes will not allow you to simply lock and unlock the device at will. There's no way to do it on iOS and Linux. The only iOS app that does this is Apple's own Watch app, and they don't expose the functionality to other apps. On Linux, some desktop environments have the ability to lock/unlock via an app, but most don't, and the only way that it _might_ be possible in a DE-agnostic way is by using root privileges which is not secure and very much not recommended, and even then, some DEs won't listen for that and the locker won't be killed or started. That's not even to mention people using standalone lockers like i3lock with a WM rather than a full DE, like I do.

Locking the device isn't usually a problem.  There's no universal standard, but there's a few widely-used ones, and HID code 19E is supposed to be a lock request.  Not sure how widely supported it is.  But in theory it shouldn't even require a companion app on the device, just the watch advertising itself as an HID sender.  People with custom configs would, obviously, have to custom configure it.  That's what you get for running a jalopy.

Unlocking the device is generally made deliberately difficult to do programmatically.  And you're right, we probably don't want to go there.

Arsen6331, 2022-06-08:
> Locking the device isn't usually a problem. There's no universal standard, but there's a few widely-used ones, and HID code 19E is supposed to be a lock request. Not sure how widely supported it is. [...]

The lock key works sometimes and sometimes does not. I don't know if it works on iOS, I think it does, but on Linux, it depends on the computer and your configuration. I happen to be the developer of a Linux companion app, which is why I am particularly concerned with that aspect.

Arsen6331, 2022-06-08:
Though just locking alone is easier, and might be possible to implement in a somewhat suitable way for all configurations, but on Linux, generally anything that depends on the OS is difficult to do unless there is a universal standard, because there are so many different configurations that essentially every single system is different.