InfiniTimeOrg/InfiniTime #1022: Inverted colors after dead battery

jgonyea, 2022-03-07:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Infinitime OS colors are inverted after running battery completely out and recharging.

### What should happen instead?

Colors should remain as normal

### Reproduction steps

I'm currently on Infinitime 1.8.0

I ran out of battery power one day when I forgot to recharge the watch.  After placing the watch on the charger, the watch rebooted, but the colors appear to be inverted (see attached photo below).  I can attach a video

The colors are correct during boot (the pine logo shows the green progress loading indicator), but as soon as Infinitime completes its boot, the colors flash the correct colors briefly, then invert.

### More details?

Attached image of inverted screen:
[image]

Attempts to fix (all fail):
* Reboot
* Reflash 1.8.0

The only fix I've found for the last time this happened is to let the watch battery completely die again, then recharge again.

### Version

1.8.0 (adc7909c)

### Companion app

Gadgetbridge 0.65.0

jgonyea, 2022-03-09:
Follow up, after letting the battery run out, this time it did not return to normal colors. The face is still inverted.

jgonyea, 2022-03-09:
Another followup:
I downgraded to 1.7.1 and the colors were correct again.  I then restarted (to roll back to 1.8.0) and colors remained normal.

jgonyea, 2022-03-09:
What's odd is that the colors are correct during the logo boot process (show a green or red pine logo), but then inverted after the full OS is booted.

geekbozu, 2022-03-09:
Could be a bad theme setting somehow... not sure off hand!

Colt45ws, 2022-03-12:
I have this problem with the PineTimeStyle watch face. One time it did this, I rebooted and it went back to normal. After the 2nd time I havent been able to fix it. I switched to Analog watch face it its normal. The menus are not affected on mine, only that one watch face.

koolfy, 2022-03-18:
Menus are affected for me when this happens, I fixed it just yesterday by charging the battery to 100%
This is all very confusing, I'm not sure I correctly describe what actually addressed the situation. 

It has happenned 3-4 times for me as well, all on 1.8.0 and I am using pinetimestyle as well (but the inverted colors happen in all menus and at a low level. I also notice boot screen is not inverted)

koolfy, 2022-04-03:
Display just inverted on me after the watch was fine, fully charged, and I unplugged it to upgrade from 1.8.0 to 1.9.0.
Right after the upgrade was finished, when the watch started 1.9.0, the bug appeared.

Maybe this helps narrow down possible causes?

lordfeck, 2022-04-27:
I notice something similar happen my PT. I think it began with IT 1.8.0 and the issue persisted after upgrading to 1.9.0.

> I'm pretty sure our screen has a mode that has color inversion. 

One way of looking at it is light mode/dark mode on the cheap!

magic-ian, 2023-09-14:
This is still present on current firmware

Alddste, 2023-11-18:
Still affects version 1.13.0

csmertx, 2023-12-04:
[image]
[image]
[image]

Barton 20mm band for breathability

Version: 1.13.0
Watch face: PineTimeStyle
Everyday user since: 21-08-09 (thank you)

Inverted colors after charging (APC surge protector 2.4A USB port)

Personally, I appreciate the aesthetic, but I can see how others may not enjoy the abrupt change

mark9064, 2025-06-28:
It seems the original issue has been resolved, and inverting at runtime is now being tracked in #1984