InfiniTimeOrg/InfiniTime #1032: Add a phone info screen

huckvrni, 2022-03-12:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Show info about the phone such as battery

### Description

Show some information about the phone, perhaps by swiping left from the time screen.
This could show phone, name, current battery (charging or not), if the phone is connected to internet and so on.
This screen could include other phone related function such as find my phone.

huckvrni, 2022-03-12:
I might look into this myself since I just got my new pinetime but I need to know if there is interest in such a thing...

0x1a8510f2, 2022-03-12:
I'm definitely interested, but this will need support from companion apps

Riksu9000, 2022-03-13:
It would be a good idea to show the "connected device" somewhere. I don't know if it needs to show much more than that though. Considering the connection range is short, you can usually just check the phone.

0x1a8510f2, 2022-03-13:
>Considering the connection range is short, you can usually just check the phone

One could say the same for notifications

huckvrni, 2022-03-13:
I would say that there is no need to flood the user with unnecessary information, there is however a need for a screen that centralizes all the functions related to the connected phone.
I also believe it shouldn't be hidden within an app, rather right next to the watch-face looks like a good spot to me.

Riksu9000, 2022-03-13:
> > Considering the connection range is short, you can usually just check the phone
> 
> One could say the same for notifications

You could out of context. It's not exactly the same though.

> I would say that there is no need to flood the user with unnecessary information

I agree with this, but disagreements can come from what is considered unnecessary. Personally I think it only needs to show the connected device. I think there needs to be a good reason and a use case before adding more information.

> I also believe it shouldn't be hidden within an app, rather right next to the watch-face looks like a good spot to me.

There have been many suggestions for what should be put on the right of the watchface. We shouldn't put anything there carelessly, because it is the last quickly accessible spot. I also wouldn't mind there never being anything, just to keep things simple.

huckvrni, 2022-03-13:
Well since it seems to be a more sensitive topic, we might need to discuss this with more people to see the most common opinion.
Another option about that side screen would be a settings to set a custom app that can be quick accessed but if so this should be discussed in another issue alltogether...

theotheroracle, 2022-03-23:
an ability to rimg a lost phone for example would be helpful