InfiniTimeOrg/InfiniTime #1035: Weather.cpp reference 'dateTimeController' is not yet bound

NeroBurner, 2022-03-13:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Clang issues a warning when compiling

### Reproduction steps

compile the project

### More details?

```
InfiniTime/src/displayapp/screens/Weather.cpp:30:25: warning: reference 'dateTimeController' is not yet bound to a value when used here [-Wuninitialized]
    dateTimeController {dateTimeController},
                        ^
1 warning generated.
```
@Avamander git blames says the line in question was written by you

It's just a debugging tool, but wanted to inform you about the usage of a reference, which isn't initialized

### Version

21da5869c5e48df666ea9d9c7698f7f69528645e

Avamander, 2022-03-13:
I'd say dateTimeController is bound when Weather debug app is instantiated the first time?

NeroBurner, 2022-03-13:
Maybe `dateTimeController` is default initialized, but initializing a variable with itself seems like a bug.

Even though it doesn't wreak havoc, because `dateTimeController` isn't used anywhere in `Weather.cpp`

JF002, 2022-03-17:
In this case, both `dateTimeController`  are `this->dateTimeController` since there is no other variables in the scope with the same name... I guess that an argument `dateTimeController` is missing in the parameter list?

It's easy to notice that with syntax highlighting:
[image]

PR #1481 (2022-12-12): Fix #1035 and unreported compiler warning.
* Fixed #1035 by removing unused member (can be added back when needed)
* Reordered member initialization in `Counter.cpp` to remove compiler warning
closed 2022-12-12 by Riksu9000