InfiniTimeOrg/InfiniTime #1041: Add companion capabilities system

Arsen6331, 2022-03-19:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

This would allow a companion app to broadcast its capabilities to the watch. Then, the watch can hide features unsupported by the companion.

### Description

I think this could be implemented by using an enum to represent the capabilities, and exposing a characteristic where a companion can send a CBOR array containing the features they support. That data can then be used when determining whether to show certain items.

This is useful because not all companions are able to support all the features, and having the button for that feature visible but doing nothing causes confusion among users and causes users to file feature requests for things that cannot be supported.

For example, ITD cannot support muting on calls because ModemManager isn't responsible for the audio and there is no way to mute a call specifically. I would have to mute the entire system, which is not ideal and would be very unreliable in terms of turning off when the ringing is over.

I do not know how feasible this is on the watch side, it would probably be a lot of work, but I think it would be a very useful feature.

huckvrni, 2022-03-20:
To add on that issue I'd say we should just implement a companion app communication protocol if one does not exist yet

Arsen6331, 2022-03-21:
> To add on that issue I'd say we should just implement a companion app communication protocol if one does not exist yet

I don't know what you mean by this. Companions communicate with the watch over BLE.

huckvrni, 2022-03-21:
I will elaborate.
I mean it in the same way a server answer http request with a known format
that the client know to parse, or how telegram has their bot api that
returns a JSON object with a known structure, drones communicate with the
mavlink protocol and so on.

What I'm going at, is that I've not seen a standard smart watch
communication protocol that any companion app can use knowing it will
support a wide range of watches out of the box, which is a shame really...

Arsen6331, 2022-03-21:
I agree that would be a nice thing to have, but I highly doubt it would be adopted in more than just open-source PineTime firmware. InfiniTime is trying to use standard protocols wherever possible (e.g. BLE FS (Adafruit BLE File Transfer), and notifications (ANS), which in my opinion is the best they can do in making it universal. I do agree that it is a shame.

Avamander, 2022-03-21:
@huckvrni

> I mean it in the same way a server answer http request with a known format that the client know to parse, or how telegram has their bot api that returns a JSON object with a known structure, drones communicate with the mavlink protocol and so on.

The Bluetooth standard specifies a lot of that, take a look at [the specs](https://www.bluetooth.com/specifications/specs/). BLE *is* the standard smartwatch communication protocol supported by many out of the box. If there isn't an existing specification for an use-case by Bluetooth SIG, then InfiniTime and many others implement protocols that are generally widespread. Last option is creating our own service with our own protocol.

huckvrni, 2022-03-21:
Well I didn't know Bluetooth had all that, pretty impressive

Arsen6331, 2022-03-21:
> The Bluetooth standard specifies a lot of that, take a look at [the specs](https://www.bluetooth.com/specifications/specs/). BLE _is_ the standard smartwatch communication protocol supported by many [...]

I think what this person meant was to create a custom protocol for all BLE services used by InfiniTime and then try to standardize them, similar to how HTTP REST APIs usually have a defined format which others can then mimic for compatibility with the original project's clients. For example, anyone can make a Matrix homeserver that works with all Matrix clients because the protocol is well-defined and standardized. While that would be nice, it isn't very practical because no one will adopt it and change how everything is implemented on their smartwatches just because some open-source project somewhere made a protocol.

tlhonmey, 2022-06-08:
@Arsen6331 What it seemed like to me was that, currently, the watch tells the app what features the watch supports, which is great, but there isn't a way for the app to tell the watch what features the app supports so the watch can hide things that won't work.

I don't know that there's a big need for this at the moment, since what few features the watch has that the app might not do still work stand-alone, but if the watch becomes more featureful in the future it could be quite useful to hide things that won't work anyway to reduce UI clutter.

But maybe the new "standard" should be as simple as "anything that only works with a partner device gets hidden unless that partner is connected and using it."  And then it's simply a matter of the watch advertising what it can do and letting the smarter device decide what to use.

Arsen6331, 2022-06-08:
> @Arsen6331 What it seemed like to me was that, currently, the watch tells the app what features the watch supports, which is great, but there isn't a way for the app to tell the watch what features the app supports so the watch can hide things that won't work.

Kind of. BLE has characteristics with UUIDs, so you can tell what features are available by looking at the UUIDs.

> I don't know that there's a big need for this at the moment, since what few features the watch has that the app might not do still work stand-alone, but if the watch becomes more featureful in the future it could be quite useful to hide things that won't work anyway to reduce UI clutter.

There are already some things (like my example, call muting)

> But maybe the new "standard" should be as simple as "anything that only works with a partner device gets hidden unless that partner is connected and using it." And then it's simply a matter of the watch advertising what it can do and letting the smarter device decide what to use.

Unfortunately, you can't do it this way, there is not an individual characteristic for every feature. If you subscribe to the call notification characteristic, you're using the whole thing, Answer, Mute, and Hang Up. This is a good thing, as having more UUIDs adds even more complexity and creates confusion. That's why my suggestion was to add a new characteristic.