InfiniTimeOrg/InfiniTime #1048: My pinetime screen changes itself

aazaya, 2022-03-23:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

I noticed my pinetime screen changing itself and sometime changes the settings or opens some apps so to reduce it i started waking it up with double tab.

### What should happen instead?

It shouldnt be changing by itself

### Reproduction steps

Whenever the screen is on randomly it changes its screen, for example, goes to some setting sometimes changing whats in the setting or playing the game by itself for a few seconds.what's

### More details?

I made a Reddit post about 5 months ago and it seems some other people are having this issue too. I used to have a wrist raise as wake up but when I was working and check time most of the time it was in a random place so I started using single tap first and now double tap. And sometimes when I wake up the device it changes to the random screen which is a real pain.
I bought pine time when it was on 1.4.0 and I always had the problem at the start I thought i was the problem but when I saw it changing few time i realized it wasn't me

[](https://www.reddit.com/r/pinetime/comments/qbaolk/screen_moves_without_touching/)

### Version

v1.8.0

### Companion app

Gadgetbridge and Siglo

aazaya, 2022-03-23:
[https://www.reddit.com/r/pinetime/comments/qbaolk/screen_moves_without_touching/](url)

Riksu9000, 2022-03-23:
Screen changing due to raise wrist wakeup method would be issue #618, and this is simply caused by accidentally touching the screen. Is your watch instead changing the screen on it's own while you're looking at it with clearly nothing touching the screen? To me it seems most likely that the touch screen on your watch is faulty, since this isn't something that most people experience.

aazaya, 2022-03-23:
sorry I forgot to mention yes it is changing by itself without me touching.

soup-bowl, 2022-06-26:
I am also experiencing this issue. I've also had my Pinetime running Wasp-OS and it did not experience the same issue.

I've seen @Riksu9000 comment about the wake-up method. I've turned them all off and will report back if the issue persists.

Riksu9000, 2022-08-01:
In case this is not the same as #618, it's a hardware defect and not an InfiniTime issue.

closed 2022-08-01 by Riksu9000