InfiniTimeOrg/InfiniTime #1049: Memory corruption when using Weather helper functions

Avamander, 2022-03-24:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Some functions in Weather are encountering corrupted local variables

### Reproduction steps

Occurs most often with current day's minimum and maximum temperature helper functions. Local variables are total garbage, writing them causes crashes.

I suspect some allocation is too small or there's an overflow. I haven't yet had the time to complete Weather emulator, but I want to do that to see if Valgrind sees something. 

### Version

develop

ITCactus, 2022-03-25:
could it be the root-cause of error response "13" on attempt to write weather characteristics?

e.g. in gadgetbridge logcat i see 
`characteristic write: 00040001-78fc-48fe-8e23-433b3a1942d0 (failed: 13)`
on every attempt to write the weather with temperature, 
or even with the simplest info - special event "Tsunami" e.g. with `81a46974696d657374616d701a623d8fe76765787069726573190e10696576656e745479706505647479706501`
that means
`[{"timestamp": 1648201703, "expires": 3600, "eventType": 5, "type": 1}]`

or that data format is simply wrong?

unfortunately, i [can't make](https://forum.pine64.org/showthread.php?tid=16031) the Pine64 USB JTAG Adapter work with  PineTime, so i can't debug the other side (or see the output, if any). and it's the first time me using cbor... 
so, sorry if it's a lame question.

Avamander, 2022-03-25:
@ITCactus It is most likely that your data format is wrong. It shouldn't be nested in a list, if I'm reading your payload correctly.

ITCactus, 2022-03-25:
@Avamander  it's array of maps (in this example, array with a single map).
i tried also without wrapping into array - just a plain map, like this
`{"timestamp": 1648201703, "expires": 3600, "eventType": 5, "type": 1}`
but the result is the same - error "13".

Avamander, 2022-03-25:
@ITCactus Ah yes, that is better, but capitalize the first letter of each key.

ITCactus, 2022-03-25:
@Avamander tried with
`{"Timestamp": 1648210579, "Expires": 3600, "EventType": 5, "Type": 1}`
but no luck.
also checked with "TimeStamp", but still the same  - error "13".

Avamander, 2022-03-25:
@ITCactus I'm honestly not sure, that seems valid. What's the binary equivalent? That error code is returned the moment one part of the payload is considered invalid. If you had a debugger you could add breakpoints to see what exactly.

ITCactus, 2022-03-25:
@Avamander  the binary for `{"Timestamp": 1648210579, "Expires": 3600, "EventType": 5, "Type": 1}` is 
`a46954696d657374616d701a623db2936745787069726573190e10694576656e745479706505645479706501`

actually, i'm using logs output like "0xa4 0x69 0x54 ..." to convert into "a46954...", and verify on https://cbor.me the decoded value.

the logs from logcat are like this:
`14:48:12.473 22970-26095/nodomain.freeyourgadget.gadgetbridge D/nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: About to run action: March 25, 14:48: WriteAction on characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0`
`14:48:12.478 22970-26095/nodomain.freeyourgadget.gadgetbridge D/nodomain.freeyourgadget.gadgetbridge.service.btle.actions.WriteAction: writing to characteristic: 00040001-78fc-48fe-8e23-433b3a1942d0: 0xa4 0x69 0x54 0x69 0x6d 0x65 0x73 0x74 0x61 0x6d 0x70 0x1a 0x62 0x3d 0xb2 0x93 0x67 0x45 0x78 0x70 0x69 0x72 0x65 0x73 0x19 0x0e 0x10 0x69 0x45 0x76 0x65 0x6e 0x74 0x54 0x79 0x70 0x65 0x05 0x64 0x54 0x79 0x70 0x65 0x01`
`14:48:12.843 22970-22996/nodomain.freeyourgadget.gadgetbridge D/nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: characteristic write: 00040001-78fc-48fe-8e23-433b3a1942d0 (failed: 13)`
`14:48:12.846 22970-22996/nodomain.freeyourgadget.gadgetbridge D/nodomain.freeyourgadget.gadgetbridge.service.btle.BtLEQueue: failed btle action, aborting transaction: 00040001-78fc-48fe-8e23-433b3a1942d0 (failed: 13)`

so, the error/status code returned by android os is literally "13".
if you refer to pinitime debugger, it doesn't work for me... and actually today, i was managed to accidentally erase (kill?) the unsealed one with nrfjprog :( ...

Avamander, 2022-03-25:
@ITCactus If you want working GadgetBridge Weather then I can provide you the source. I have it done but not polished up.

ITCactus, 2022-03-25:
@Avamander if you implemented the weather sync in gadgetbridge, why not to add a pull-request to the gadgetbridge repo (even WIP) for feature-request [#2546](https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2546) ?
for sure i would be interested to take a look at code. because looks like i'm stuck on this several month ago... well, ok, i don't have enough time to work on this, and all the progress on attempts was finished 2 month ago, and meanwhile i tried to connect the Pine64 USB JTAG Adapter and make it work with PineTime... and now, i have not worked unsealed PineTime... so... the progress would be even slower, if any...
but i'm really wondering, why my code doesn't work... maybe the issue is in the "co.nstant.in.cbor.CborEncoder".

Avamander, 2022-03-29:
@ITCactus 

> If you implemented the weather sync in gadgetbridge, why not to add a pull-request to the gadgetbridge repo (even WIP) for feature-request [#2546](https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2546) ?

Because it's not finished. I've only given Kieran a test build.

closed 2023-04-30 by Avamander