InfiniTimeOrg/InfiniTime #1055: Screen overflow if more than four watch faces

hrmckay, 2022-03-27:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

On Watch Face selection screen entries flow off the bottom

### What should happen instead?

Watch Face selection screen should scroll

### Reproduction steps

In files SettingWatchFace.cpp and SettingWatchFace.h adding another watch face after the four existing watch faces causes the fifth watch face to show only half the name and button at the bottom of the screen.  A sixth watch face would not show at all.  In terms of code changing:
static constexpr std::array<const char*, 4> options = {" Digital face", " Analog face",  " PineTimeStyle", " Terminal"}; 
to:
static constexpr std::array<const char*, 5> options = {" Digital face", " Analog face",  " PineTimeStyle", " Terminal", " Weather face"};
causes "Weather face" to only show half the text and button at the bottom of the screen.

### More details?

Even though the fifth watch face button and name are only half shown on the bottom of the screen it is still selectable.  I have done nothing to try and fix it.

### Version

1.8.0

explit7, 2022-03-29:
I would be very happy to have this Watchface added by default:
https://zephyrlabs.github.io/Watchfaces/Digistyle/
[image]

The Problem is the Name of the Watch. Its name is also DigiStyle , which interfere with default Digistyle Watchface. Is it possible to add it under another Name ?

Thanks

NeroBurner, 2022-04-03:
For reference: In the Infineat PR https://github.com/InfiniTimeOrg/InfiniTime/pull/1024/files#diff-9a831e18eb55ca2a71cd4c049332c67dcb0d7607109ebddf043757e5276b4f43 @dmlls created a `CheckboxList.h` class which solves this problem

Tiggilyboo, 2022-05-22:
> For reference: In the Infineat PR https://github.com/InfiniTimeOrg/InfiniTime/pull/1024/files#diff-9a831e18eb55ca2a71cd4c049332c67dcb0d7607109ebddf043757e5276b4f43 @dmlls created a `CheckboxList.h` class which solves this problem

Started to look into this as well for the [step meter analog face](https://github.com/InfiniTimeOrg/InfiniTime/pull/1151). Wouldn't something like the [lv_roller](https://docs.lvgl.io/latest/en/html/widgets/roller.html) widget be better suited (and not have any complexity with the multiple screens and control handling? I think the CheckBoxList will work well for many multi-selections, but in the case of radio button (single selections) I think it's a bit overkill...

Something to discuss and see what people think?

JF002, 2022-08-02:
I've just created a more generic feature request about dynamic app and custom watchface to group multiple requests around that topic. Feel free to continue the discussion in #1262 ;-)

closed 2022-08-02 by JF002