InfiniTimeOrg/InfiniTime #1056: Linking not working on Windows 10

watertrainer, 2022-03-27:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

When trying to build with Windows 10 after following [buildInVsCode.md](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/buildWithVScode.md) on WIndows, everything builds but the linking fails and throws an error.

### What should happen instead?

Linking should work and building complete without error

### Reproduction steps

On Windows 10 follow [buildInVsCode.md](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/buildWithVScode.md) and [buildAndProgram.md](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/buildAndProgram.md).

I used cmake 3.19.4 and make 3.81, installed with mingw64 and gnuwin32 respectively.
When trying to build using ```make -j pinetime-app``` it builds to 100% and then fails on the following message.
```c:/program files (x86)/gnu arm embedded toolchain/9 2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: unrecognized option '--major-image-version'
c:/program files (x86)/gnu arm embedded toolchain/9 2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: use the --help option for usage information      
collect2.exe: error: ld returned 1 exit status
make[3]: *** [src/pinetime-app-1.8.0.out] Error 1
make[2]: *** [src/CMakeFiles/pinetime-app.dir/all] Error 2
make[1]: *** [src/CMakeFiles/pinetime-app.dir/rule] Error 2
make: *** [src/CMakeFiles/pinetime-app.dir/rule] Error 2```

### More details?

I tried to remove the argument in question and found, that ld.exe is called in [CMakeList.txt:913](https://github.com/InfiniTimeOrg/InfiniTime/blob/a8b7fbe48b4a86238f38ed0f084b277b44c428fb/src/CMakeLists.txt#L913). There doesn't seem to be an argument ```--major-image-version``` being passed to ld.exe.

When I try it in linux, everything works, although the counterpart to ld.exe doesn't accept and ```major-image-version``` argument either, even though it shows up on the man page.

### Version

newest Version, development Branch

Avamander, 2022-03-28:
Generally Windows is not a supported setup, but if you find a fix then maybe it can be documented.

ITCactus, 2022-03-28:
@watertrainer not sure if this is the case, but in my experience, spaces in the path can cause a lot of issues. the second one - a looong path. 
so, maybe creating a sym-link (mklink) to desired folder with a short link e.g. "c:\arm-none-eabi" so it will end up with "c:\arm-none-eabi\bin\ld.exe" and use it in PATH.

in short, i'm also on Win10. 
and ended up in using VS Code/Docker DevContainer when VS Code suggested.
you will need later to install cbor with `pip install cbor` in terminal on container to make DFUs,
and change EOL symbols (e.g. with Notepad++) from Windows to Linux format on 5 python scripts:
- \InfiniTime\tools\mcuboot\imgtool.py
- \InfiniTime\tools\mcuboot\imgtool\__init__.py
- \InfiniTime\tools\mcuboot\imgtool\boot_record.py
- \InfiniTime\tools\mcuboot\imgtool\image.py
- \InfiniTime\tools\mcuboot\imgtool\version.py

and that's all.

watertrainer, 2022-03-29:
Sadly none of the suggestions worked for the linking, I'll probably go for Docker then as well. 
I think it might be helpful to mention in the docs, that for windows natively building without some kind of linux virtualization isn't working at the moment.

Thanks for the tip later down the line, I'll keep that in mind!

watertrainer, 2022-04-07:
There is now a pull request to document what needs to be done to build on windows, so I think this issue can be closed.

closed 2022-04-07 by watertrainer