InfiniTimeOrg/InfiniTime #1058: Feature Request: InfiniPaint save/load image, BT send/receive image and "Make Wallpaper" function

Danimations, 2022-03-29:
I would love to see the ability to see the following features added to Infinitime:

1) save and load artworks made in InfiniPaint
2) send/receive .bmp files of 240x240 px resolution via Bluetooth
3) assign a .bmp image to be a clockface background

I've just upgraded to Infinitime 1.8.0 and am very impressed with how the OS is progressing. I'm a designer/animator with a focus on low-res pixel art, so if there's anything I can assist this project with, just holler.

Avamander, 2022-03-31:
BMP as a format is out of the question because it's very incompatible with our hardware limitations.

Danimations, 2022-03-31:
Which image format would be the most compatible/implementable @Avamander ?

Avamander, 2022-03-31:
@Danimations anything that follows the requirements specified here: https://docs.lvgl.io/latest/en/html/overview/image.html#image-decoder

tigercoding56, 2023-02-17:
what if we just stored the image as a series of movements the user made with the brush? , kind of like a mini svg format ?