InfiniTimeOrg/InfiniTime #1059: Please add zephyrlabs Digistyle Watchface by deafult.

explit7, 2022-03-29:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

zephyrlabs Digistyle Watchface

### Description

I would be very happy to have this Watchface added by default:
https://zephyrlabs.github.io/Watchfaces/Digistyle/
[image]

The Problem is the Name of the Watch. Its name is also DigiStyle , which interfere with default Digistyle Watchface. Is it possible to add it under another Name ?

Thanks

explit7, 2022-03-29:
@ZephyrLabs

ZephyrLabs, 2022-03-30:
regd the watchface name; It could be renamed to casiostyle or oldstyle, its purely what the general consensus is to name it :)

regd the build issues; it needs to be implements just like any other watchface, currently busy at the moment, maybe @kieranc can provide some guidance

ITCactus, 2022-03-30:
> It could be renamed to casiostyle or oldstyle

maybe casiostyle classis?...

i would like to see also the casio style like that from [this thread
](https://forum.pine64.org/showthread.php?tid=16279&pid=108016#pid108016) with week number, day number and days until the end of the year...
maybe after 1.9 release will UI emulator i'll find some time to implement it...

[image]

explit7, 2022-03-30:
yes, CasioStyle would be good...

explit7, 2022-03-30:
I am not a developer, maybe someone can point me to the right direction here:

`[ 36%] Building CXX object src/CMakeFiles/pinetime-app.dir/displayapp/screens/WatchFaceDigital.cpp.o
/media/sdc1/Projects/InfiniTime-Builder/InfiniTime/src/displayapp/screens/WatchFaceDigital.cpp: In member function 'virtual void Pinetime::Applications::Screens::WatchFaceDigital::Refresh()':
/media/sdc1/Projects/InfiniTime-Builder/InfiniTime/src/displayapp/screens/WatchFaceDigital.cpp:130:63: error: no matching function for call to 'Pinetime::Applications::Screens::BleIcon::GetIcon(const bool&)'
  130 |     lv_label_set_text(bleIcon, BleIcon::GetIcon(bleState.Get()));
      |                                                               ^
In file included from /media/sdc1/Projects/InfiniTime-Builder/InfiniTime/src/displayapp/screens/WatchFaceDigital.cpp:7:
/media/sdc1/Projects/InfiniTime-Builder/InfiniTime/src/displayapp/screens/BleIcon.h:10:28: note: candidate: 'static const char* Pinetime::Applications::Screens::BleIcon::GetIcon(bool, bool)'
   10 |         static const char* GetIcon(bool isRadioEnabled, bool isConnected);
      |                            ^~~~~~~
/media/sdc1/Projects/InfiniTime-Builder/InfiniTime/src/displayapp/screens/BleIcon.h:10:28: note:   candidate expects 2 arguments, 1 provided
/media/sdc1/Projects/InfiniTime-Builder/InfiniTime/src/displayapp/screens/WatchFaceDigital.cpp:181:25: error: 'displayedChar' was not declared in this scope; did you mean 'displayedHour'?
  181 |     if (hoursChar[0] != displayedChar[0] || hoursChar[1] != displayedChar[1] || minutesChar[0] != displayedChar[2] ||
      |                         ^~~~~~~~~~~~~
      |                         displayedHour
make[2]: *** [src/CMakeFiles/pinetime-app.dir/build.make:790: src/CMakeFiles/pinetime-app.dir/displayapp/screens/WatchFaceDigital.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:252: src/CMakeFiles/pinetime-app.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
`

ZephyrLabs, 2022-03-31:
send a link to the source if it is on github, it will help to debug it :slightly_smiling_face:

explit7, 2022-03-31:
> send a link to the source if it is on github, it will help to debug it slightly_smiling_face

The source of the file is not on Github, its on your Github Pages.
https://zephyrlabs.github.io/Watchfaces/Digistyle/

ZephyrLabs, 2022-03-31:
the code written for that watchface is quite outdated, considering how much stuff has changed in the WatchFaceDigital.cpp file, it obviously wouldn't compile as DisplayedChar is not even declared in the header or the source file, to move the old watch face to the new file it will take a couple steps to modify and adapt to the new setup

explit7, 2022-03-31:
Thanks for the info, @ZephyrLabs Maybe you or someone else will find time to rebuild it...

Danimations, 2022-04-20:
I am just writing to support the bid to have this watchface included with the next Infinitime version. I like the nod it gives to the history of digital watches.

explit7, 2022-04-22:
> I am just writing to support the bid to have this watchface included with the next Infinitime version. I like the nod it gives to the history of digital watches.

Thanks, very cool! I wait for it!

explit7, 2022-04-29:
Wow, it looks really great! Yes i know the problem with the size. I think it was the case, why @ZephyrLabs has published it instead the origininal Digistyle Watchface. Maybe you could fork it and build it instead the original Digistyle ?
Thank you very much, @ITCactus !

kieranc, 2022-04-30:
> [image] 

This looks super cool, shame it's a bit too big. I'm guessing you're using an image for the background and top/bottom text, could it be split into smaller images or replaced with lvgl code? It wouldn't looks quite as good but it would maybe fit on the watch.

explit7, 2022-04-30:
Well, i have not designed the watchface, but maybe @ZephyrLabs has, and he can answer the question... My all time favorite watchface. One love.

ZephyrLabs, 2022-04-30:
Hey guys,
sorry regarding the image not being able to fit on the internal flash rom, 
there is a fix to this which is to use indexed image with less bpp (bits per pixel) this will reduce the usage and allow it to fit with in the rom space,

the current one provided in the src is 16 colors indexed or 4 bpp, I have taken the pleasure to convert it to a 1bpp indexed version it linked to this comment, this image should be able to fit in the flash without any issue, cheers :)
[bitmaps.zip](https://github.com/InfiniTimeOrg/InfiniTime/files/8596725/bitmaps.zip)
)

explit7, 2022-04-30:
@kieranc Could you try to build it with @ZephyrLabs bitmaps? That would be awesome. To fork your build somewhere here would be also great, until the watchface will its way to the master branch....
Thanks guys!

JF002, 2022-04-30:
This watchface looks really awesome, indeed! Thanks for porting it to the 1.9 codebase.

Memory usage is the main difficulty right now, and we have to think very carefully before merging features that use a significant amount of memory to avoid filling RAM and/or ROM too much.
One of the next priorities for the project will be to enable the external flash memory and the filesystem so we can store big bitmaps and fonts on it. This should allow us to merge new watchfaces and apps ;)

ITCactus, 2022-05-02:
> I have taken the pleasure to convert it to a 1bpp indexed version

now it fits even in DEBUG build:
> [build] Memory region         Used Size  Region Size  %age Used
[build]            FLASH:      482688 B       480 KB     98.20%
[build]              RAM:       54528 B        64 KB     83.20%

and regarding #1055, i see [this comment](https://github.com/InfiniTimeOrg/InfiniTime/issues/1055#issuecomment-1086929012), so i'll try to add it as a 5th watchface ("Casio Digital"), so it will not conflict with existed "Digital face".

i understand, that this will not be accepted into the mainline, 
but i'll try to make PR with attached build for OTA update based on InfiniTime 1.9 release.

explit7, 2022-05-02:
@ITCactus Could you please share updated files from this watchface, so we can build own firmware with Infinitimebuilder, until it takes time to find its way to the build?

or maybe just share the built firmware?

this would be cool.

Thank you for your work!

ITCactus, 2022-05-02:
i prepared the updated version of digistyle-Modpack for InfiniTime 1.9
not sure, if it will work with [InfiniTime-Builder](https://github.com/ZephyrLabs/InfiniTime-Builder),
but it contains updated code and 1bpp image (compatible with InfiniTime 1.9 "master" branch):
- src\displayapp\screens\WatchFaceDigital.cpp
- src\displayapp\icons\digistyle.png
- src\displayapp\icons\digistyle.c
- src\CMakeLists.txt

font files are unchanged.

the result file
[digistyle-Modpack-InfiniTime-1.9.zip](https://github.com/InfiniTimeOrg/InfiniTime/files/8602569/digistyle-Modpack-InfiniTime-1.9.zip)

explit7, 2022-05-02:
Awesome!