InfiniTimeOrg/InfiniTime #1060: Keeping bluetooth off

WhyNotHugo, 2022-03-30:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Keep bluetooth off as much as possible

### Description

- When the clock turns on, try to connect to a trusted device for a while.
- Once one is found, sync the time, disconnect, and keep bluetooth completely off.

I switch in `settings` to turn it back on would be useful, for updating or other manual tasks. Something like "bluetooth radio": auto/on/off. `auto` does what I've  described above, `on` keeps it on permanently, `off` doesn't turn it back on after running out of battery and rebooting.

Maybe instead of labeling it "bluetooth radio" just put this icon:

[image]

Elements6007, 2022-04-01:
i think that it will be implemented in #888 with airplane mode

Danimations, 2022-04-03:
I also support this feature request, for battery and privacy reasons.

JF002, 2022-04-12:
In case you haven't noticed yet : we released [InfiniTime 1.9](https://github.com/InfiniTimeOrg/InfiniTime/releases/tag/1.9.0) a few days ago. This new version offers the possibility to manually turn the BLE radio ON and OFF. It's available in the setting menu (3rd screen).

With this feature, BLE is enabled when the watch boots. It'll then connect to your companion app if it is in range. Then, you'll be able to disable BLE if you don't need it and manually re-enable when you need it for an update, for example.

Knowing this, do you think that an auto off mode is still needed? 

Now, if we wanted to implement this "auto-off" feature:
 - what would be the ideal delay before automatically shutting down the radio? 30s? 5 minutes? Not that I would prefer we avoid adding a setting to specify that delay.
 - How would the user re-enable the bluetooth temporarily when the "auto-off" mode is enabled?

WhyNotHugo, 2022-04-12:
I did notice 1.9 and I'm using it now, it's a very welcome upgrade!

> what would be the ideal delay before automatically shutting down the radio? 30s? 5 minutes? Not that I would prefer we avoid adding a setting to specify that delay.

Disabling it after the time has been synced would be ideal. I really only _ever_ want bluetooth when I need to re-sync the time.

> How would the user re-enable the bluetooth temporarily when the "auto-off" mode is enabled?

I envisioned bluetooth as a three-way setting:

- on: keeps it on forever (e.g.: the pre-1.8 behaviour).
- off: keeps it off forever (even after rebooting).
- auto: only turns it on when booting, and turns it off once the time has been synced again.

Not sure if `auto` is the best label for it, but I also can't think of anything better. I think the current behaviour is actually somewhere between `auto` and `off`, since, as far as I can tell, it does turn on when rebooting. It just doesn't turn off after that.

UsernameRandomlyGenerated, 2022-04-16:
> In case you haven't noticed yet : we released [InfiniTime 1.9](https://github.com/InfiniTimeOrg/InfiniTime/releases/tag/1.9.0) a few days ago. This new version offers the possibility to manually turn the BLE radio ON and OFF. It's available in the setting menu (3rd screen).
Personally, I think that bluetooth toggle should be more accessible than by going to settings menu. I think being able to turn it on/off just by tapping on the bluetooth icon after a swipe from main screen would be ideal, though I guess the whole menu should be reorganized then, as currently 4 icons are taking all the space. Currently to enable/disable bluetooth you need to swipe right, go to settings, scroll down, tap on bluetooth, turn it on/off there and go back to main screen. I believe there are more people like me who would turn it on or off on a regular basis, for example enabling bluetooth for work to get notifications but disabling it at home. 

Though as I read @WhyNotHugo posts I think having something like "bluetooth auto enable-disable planner" would also be nice, basically you'd set up time at which bluetooth would be turned on and turned off automatically, basically something like "do not disturb" plan.

JF002, 2022-04-30:
I agree with @Riksu9000 : Those use-cases seems very specific (enable/disable BLE at specific time, with specific timeout,...). I don't think these behaviors will be easily understood by most of the users.
Persisting the BLE enable/disable option is however probably a good idea!

WhyNotHugo, 2022-05-01:
The big problem I'm trying to solve is the watch not having time after running out of battery. At that point, I'll put it in the charger. I won't pick it up again until I'm about to leave home:

If bluetooth was always **off*, then the watch will have no time, and I'll be realising this as I'm stepping out of home and and stuck with a timeless watch.

The alternative is to have bluetooth **on**, but that drains way more battery permanently.

What I'm looking for mostly, is for bluetooth to automatically turn on when recovering from "out of battery", but turn off once it has a time. I mostly wear it when going outside, so I only notice this at a very inconvenient moment.

That said, the setting itself doesn't need to be a third option in the bluetooth menu; it can be somewhere else instead. Like a "automatic bluetooth when timeless" checkbox or something.

WhyNotHugo, 2022-05-01:
I should also say that, now that bluetooth can be turned off manually, battery has been **a lot** better and running out of battery has been a lot less frequent.

Riksu9000, 2022-05-01:
It's unfortunate that you've been having this issue, however I don't think letting the watch run out of battery is a use case that should be supported. Instead we should make improvements to the battery life and the percentage reporting. Note that currently when the percentage is reported at 40%, it may in reality be almost empty, which can lead to accidentally letting the battery run dry.

closed 2023-02-22 by Riksu9000