InfiniTimeOrg/InfiniTime #1061: Metronome: tap tempo to establish BPM

Danimations, 2022-03-31:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Expanded metronome app: tap a rhythm on the smartwatch to determine its BPM

### Description

The metronome app on Infinitime 1.8.0 is already quite versatile, but as a DJ and musician, there's another metronome function that I'd find extremely useful. I'd like to have the ability to tap on the smartwatch screen, and have the smartwatch determine what the tempo/BPM of my tapping is.

Riksu9000, 2022-03-31:
This is actually something you can already do by tapping on the middle of the screen.

closed 2022-03-31 by Riksu9000

Danimations, 2022-03-31:
Brilliant! Many thanks.

Danimations, 2022-03-31:
After experimenting with this feature, I'd like to suggest a potential improvement. While the tempo-tapping in its current form works, a better result could be achieved if a sample of multiple taps was made, then averaged. For example, the app could measure the intervals between 8 consecutive taps, dive that by 8 and resolve a BPM. The number 8 may not be ideal for non 4/4 time signatures, but it's probably a good starting point as it will supply enough data to improve the accuracy of the end result.

What do you think, @Riksu9000 ?

Riksu9000, 2022-03-31:
That's certainly a good idea if someone wants to implement that.