InfiniTimeOrg/InfiniTime #1063: Enhanced colour picker for watch-face customisation

Danimations, 2022-04-01:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Extended palette colour picker and save/load "favorites" for PTS watch-face customisation

### Description

I love that I can change the colours of the watchface elements in the PTS watchface, but I find the currently available palette very limiting. As of v.1.8.0 there are only 16 colours (and no orange).

Ultimately, it would be great if a user could do any of the following:

1) pick a colour by dragging a pointer around a continuous gradient palette, as per Photoshop and similar programs.
2) assign a colour by hex code
3) choose from a larger palette of fixed colours (a set of 32 colours would be desirable)
4) save and load particular "favorite" colour combinations

Considering option #3, a very nice 43 colour palette based on the official colours of Lego blocks, with hex codes, is available here, : https://lospec.com/palette-list/lego-brick-2016-palette

mjhershfield, 2022-04-02:
Hello @Danimations!
I've been working on a PR (#1030) for making a global color scheme and color picker for Infinitime. I haven't fully updated PTS to work with my system, but I just made a settings page for selecting and tinting two accent colors and a button color that appear throughout the UI. I just posted some zip files on my most recent comment that you can upload to your Pinetime if you would like to try out the extended color palette and theming. At the moment, [material-colors-LUT.zip](https://github.com/InfiniTimeOrg/InfiniTime/files/8401370/material-colors-LUT.zip) has a larger selection of colors, but many of them don't look good on the PineTime's screen. If you give one of the color picking systems a try, I would appreciate any feedback you have on how I could improve them. Thanks!

Danimations, 2022-04-02:
Thanks @mjhershfield , great initiative. Could you give me more specific directions re: sending your work to my phone? Does this augment the existing firmware or replace it, and is there any risk of this bricking my watch? I only have a sealed Pinetime unit at present, and thus far have only sent an infinitime firmware update to it.

mjhershfield, 2022-04-02:
The zip is a normal firmware update. You can flash it as you would flash a standard Infinitime update. It has been running on my sealed Pinetime for two days now with no issues, so I believe the risk of bricking your Pinetime is low.

Danimations, 2022-04-06:
Thanks @mjhershfield ... since Infinitime 1.9.0 was released this week, could you please update your firmware so that I can keep the features of 1.9.0 while testing your colour picker? Thanks in advance.