InfiniTimeOrg/InfiniTime #1064: More noticeable vibration when countdown timer reaches zero

Danimations, 2022-04-02:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Adding a more noticeable attention-seeking vibration & screen flash when countdown timer reaches zero

### Description

I used the countdown timer for the first time today while cooking. I missed the simple vibration at the end of the countdown, which resulted in a burnt meal and some cursing. As the device should actively be seeing the wearer's attention at the end of the countdown, it might pay to enhance the vibration with some changes.

Firstly, I think the countdown timer would be more effective if it used the **higher-pitched vibration** (as used by the metronome app for beats 2-4 in a standard 4/4 time bar)... instead of the lower pitched one. This should make the vibration more audible in many settings.

Another consideration would be for the **vibration to buzz two or three times**, which would be harder for the wearer to miss, if he/she is not paying attention. It could also be indicated by a **flash of the screen's background to white (like a torch flash)**. It would be a nice touch if the flash or flashes syncronised with the vibration(s).

Overall the timer app is excellent. Easy to set and clear to read.

Thanks again for another useful feature.

SteveAmor, 2022-04-02:
What version do you have? The official V1.8.0 has the timer start counting back up and vibrating once a second after reaching zero. The text is also red.  It appears to be broken in subsequent (CI) builds and I can't even build V1.8.0 myself without it vibrating once on zero and stopping. Consequently I've not been able to find out why it's no longer counting up and vibrating once a second.

Danimations, 2022-04-03:
I'm running 1.8.0. In my case, the timer hit zero, gave one weak vibration, timer remained at zero, text remained grey, and no further vibrations followed. Sounds like you've already solved the problem I've suggested a fix for.... but the version in the current official build may have been rolled back in error?

Riksu9000, 2022-04-03:
That feature hasn't been in a release. I think you're talking about PR #557, which is not merged.

Also I'll close this as a duplicate of #391.

closed 2022-04-03 by Riksu9000