InfiniTimeOrg/InfiniTime #1066: 1.9.0 update from 1.8.0 gets stuck (Connection failed (0x85): GATT ERROR)

lytex, 2022-04-02:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

The upgrade from 1.8.0 to 1.9.0

### What should happen instead?

The device should upgrade to 1.9.0

### Reproduction steps

Upgrade from 1.8.0 to 1.9.0, either using the .zip or the .bin app image using nRF Connect. The update always gets stuck at 13%.

### More details?

nRF Connect, 2022-04-03
InfiniTime (ED:4D:29:44:14:32)
V 00:22:13.389 Connecting to ED:4D:29:44:14:32...
D 00:22:13.389 gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D 00:22:13.740 [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D 00:22:13.753 [Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I 00:22:13.753 Connected to ED:4D:29:44:14:32
V 00:22:13.762 Discovering services...
D 00:22:13.762 gatt.discoverServices()
I 00:22:14.069 Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D 00:22:15.333 [Callback] Services discovered with status: 0
I 00:22:15.334 Services discovered
V 00:22:15.365 Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Device Information (0x180A)
- Manufacturer Name String [R] (0x2A29)
- Model Number String [R] (0x2A24)
- Serial Number String [R] (0x2A25)
- Firmware Revision String [R] (0x2A26)
- Hardware Revision String [R] (0x2A27)
- Software Revision String [R] (0x2A28)
Current Time Service (0x1805)
- Current Time [R W] (0x2A2B)
Unknown Service (00000000-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [N] (00000001-78fc-48fe-8e23-433b3a1942d0)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [R W] (00000002-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00000004-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00000003-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00000005-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00000006-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00000007-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00000007-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00000008-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00000009-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (0000000a-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (0000000b-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (0000000c-78fc-48fe-8e23-433b3a1942d0)
Unknown Service (00040000-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [W] (00040001-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00040002-78fc-48fe-8e23-433b3a1942d0)
Unknown Service (00010000-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00010001-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00010002-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00010003-78fc-48fe-8e23-433b3a1942d0)
- Unknown Characteristic [R W] (00010004-78fc-48fe-8e23-433b3a1942d0)
Alert Notification Service (0x1811)
- New Alert [W] (0x2A46)
[... 255 lines cut ...]
E 00:23:09.937 [DFU] Device has disconnected
D 00:23:09.938 [DFU] gatt.disconnect()
D 00:23:09.950 [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
D 00:23:09.967 [DFU] gatt.close()
D 00:23:09.967 [DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
V 00:23:10.013 [DFU] DFU service started
I 00:23:10.013 [DFU] Firmware file opened successfully
D 00:23:10.013 [DFU] wait(1000)
D 00:23:11.001 [DFU] wait(1000)

Julian, [3/4/22 0:23]
V 00:23:11.990 [DFU] Connecting to DFU target...
D 00:23:12.020 [DFU] gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferredPhy = LE_1M | LE_2M)
E 00:23:17.040 [DFU] Connection failed (0x85): GATT ERROR
W 00:23:17.040 [DFU] Retrying...
D 00:23:17.040 [DFU] gatt.refresh() (hidden)
D 00:23:17.040 [DFU] gatt.disconnect()
D 00:23:17.043 [DFU] gatt.close()
V 00:23:17.062 [DFU] DFU service started
I 00:23:17.062 [DFU] Firmware file opened successfully
D 00:23:17.062 [DFU] wait(1000)

### Version

1.8.0

Joshndroid, 2022-04-02:
Mine doesn't even get that far.
I tried with gadgetbridge and it says "Element cannot be installed"

I downloaded and installed NRF Connect
Mine gets to 0%, reboots (3 times total) with it getting to 0% and fails out.

Avamander, 2022-04-02:
Have you tried something besides NRF Connect?

huckvrni, 2022-04-03:
I tried the dfu zip and gadgetbridge twice and everything worked fine

sssmack, 2022-04-03:
Siglo did not work for me (using that from the pinephone - non Pro)
Siglo reported "The firmware update has failed. Retrying the update will fix it in most cases".
I re-tried about 20 times with no change in response from Sigo.
No whining from me here. :)  The purpose of my post is to provide more info. to you - whom I greatly appreciate your time and effort on the software.  My have been letting my regular job and family take my time.

Avamander, 2022-04-03:
Maybe it's worth changing the ambient radio environment, might be noise that causes the uploads to fail.

Rebooting all components in the setup as well, but you've probably done that.

Joshndroid, 2022-04-03:
> Maybe it's worth changing the ambient radio environment, might be noise that causes the uploads to fail.
> 
> Rebooting all components in the setup as well, but you've probably done that.

Moved to another part of the house sadly the same result... Also have been rebooting like a mad man as well as repairing device multiple times. Can't even flash the same firmware thats on it, let alone anything else.

sssmack, 2022-04-03:
I changed the ambient radio environment and rebooted both devices .  
After doing that, the download and install worked the 1st time and as well as the previous downloads and installs.
Thank you!

huckvrni, 2022-04-03:
Have any of you had similar issues before 1.8?

Joshndroid, 2022-04-03:
After 6 attempts using my PinePhonePro and Siglo on Manjaro ARM i was able to get it flashed. Never had an issue with my android phone in the past. Kinda weird but yeah

huckvrni, 2022-04-03:
I think this might have to do with previously configured secure pairing in ble

huckvrni, 2022-04-03:
did any of you configured secure pairing?

Joshndroid, 2022-04-03:
Is that the "companion device pairing' in gadget bridge? I had tried that both off and on if so

huckvrni, 2022-04-03:
I have a suspicion it might have to with this I will check on my PC soon

huckvrni, 2022-04-03:
I'm seeing a lot of strange changes in [NimbleController](https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/src/components/ble/NimbleController.cpp) namely in [line 189](https://github.com/InfiniTimeOrg/InfiniTime/blob/b498e1d633522eed975d78b04508834b7a79befe/src/components/ble/NimbleController.cpp#L189) and the gav event switch but also in general, see [full diff](https://github.com/InfiniTimeOrg/InfiniTime/compare/1.8.0...1.9.0#diff-c69ced976eada3d4c4a15fd5ce542651ff381e21e2bb8cee746ac198c8847f89).

I can't investigate further at the moment...

Strit, 2022-04-03:
I have this issue coming from version 1.7.1 too. Tried both GadgetBridge and nRF Connect on Android.

marcdw1289, 2022-04-03:
Failing miserably here with Gadgetbridge. Mostly fails at about 5%, few times into the 30s, only once up to 75%.
Frustrating. Looking forward to seeing that terminal watchface. 😁

lytex, 2022-04-03:
I got mine working installing with gadgetbridge from the .bin. I initially moved from it to nRF Connect because it was gadgetbridge which got stuck.

guek, 2022-04-04:
* Gadgetbridge V0.66.0, Comm 3423dbde9
* nRF Connect V4.25.3
* Android V8.1.0
* Upgrade InfiniTime 1.8 ->1.9

No succes with nRF Conn. (0% - reboot, x3, see https://github.com/InfiniTimeOrg/InfiniTime/issues/1066#issuecomment-1086740627)

**Success** by first attempt with **Gadgetbridge** and dfu-zip.
I could provide some log from nRF. I suspect something changed with their new version.

HTH

StarGate01, 2022-04-04:
My update experience is hit and miss as well, updates via NRF connect on Android stopped working some time ago, but that might be due to an update of the app. Siglo sometimes manages to update. I suspect that the watch can't cope with some (presumably) higher datarate? 

The only tool that works somewhat reliably for me (>50% of tries) is the `bootloader/ota-dfu-python/dfu.py ` script.

hrmckay, 2022-04-04:
The update of my dev kit PineTime from 1.8.0 to 1.9.0 worked first time.  Pixel 3A phone, Android 12, Gadgetbridge 0.65.  I used the .zip file.

Joshndroid, 2022-04-04:
Bizarre that there is so much variation in updating, whats working, whats not and how many times its taking various persons to get the update to stick.

huckvrni, 2022-04-05:
Did anyone try to downgrade to previous versions?
This might shed some light on the matter

Joshndroid, 2022-04-05:
> Did anyone try to downgrade to previous versions? This might shed some light on the matter

So i didn't bother with downgrading but I got the same error when trying to just reflash the same version I was on.

Riksu9000, 2022-04-05:
Use the dfu .zip file when flashing with most companion apps like Gadgetbridge or nRF Connect. The image .bin file is only used by Amazfish. **If the firmware upgrade fails, restart your PineTime before retrying.**

I upgrade the firmware with Gadgetbridge often. I used to have some issues a long time ago, but nowadays I never have any issues. Same thing with reconnecting. I tried using Siglo, and couldn't get it to work. I had trouble getting it to detect the watch. I had it fail at 0%, at 100%, and now it won't even start.

From my experience, all firmware upgrades after a single failed attempt will also fail most of the time, if not always, until the watch is restarted. If the upgrade fails, please restart your watch by holding down the button until you see the pine logo.

I'm doubtful that there's any new issue with InfiniTime. People have reported issues with all versions.

guek, 2022-04-06:
> Did anyone try to downgrade to previous versions? This might shed some light on the matter

I tried to reflash V1.9: No success with nRF and GB.
Same with downgrade to 1.8: Both not working.
After several attempts only 1 reflash of V1.8 was sussessful.

It's no fun.

StarGate01, 2022-04-06:
I had better luck when I moved the watch closer to the Bluetooth adapter / phone, which seems to indicate that dropped packets could be a problem. Maybe I have a lot of RF interference.

Strit, 2022-04-09:
I switched to try Siglo. Worked like a charm the first time. So the issue is likely not related to InfiniTime itself, but rather the connection between the phone and the watch.

loidinhm31, 2022-04-09:
nRF restart my device many times, only works with gadgetbridge

JF002, 2022-04-10:
Remember that nRFConnect is a proprietary application by Nordic Semiconductor. They might have pushed an update that makes it incompatible with the DFU implementation in InfiniTime.

I strongly recommend using one of the open source [companion app](https://github.com/InfiniTimeOrg/InfiniTime#companion-apps) as they implemented the DFU protocol specifically for InfiniTime.

Also, note that the OTA is some kind of stress tests for both the phone/computer side and InfiniTime : it transmits a lot of data in a short amount of time. A single issue during the transfer will make the OTA fail. It can be a packet lost or corrupted, RF interference, bug in the BLE stack of the phone or in InfiniTime,...

So, as other said, reboot your watch between every attempts, try with another phone/computer and another companion app and, if possible, analyze and debug to find out why the transfer failed and how we could improve the code ;-)

advation, 2022-04-13:
What worked for me (Android 12) was to stop using nRFConnect and connected via GadgetBridge 0.66, downloaded the .zip, and then in a file browser opened the .zip with GadgetBridge. This worked first time without incident.

Q-lds, 2022-04-15:
Same here: couldn't really use nRFConnect (4.24.3) as it always got stuck at the 0% stage, while GadgetBridge (0.59.2) worked at the first attempt. Upgrading from 1.2 to 1.9

closed 2022-06-12 by Riksu9000