InfiniTimeOrg/InfiniTime #1071: Option to disable seconds display in new watchface (Terminal) in firmware 1.9.0

ajack2001my, 2022-04-05:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Remove seconds in Terminal watchface

### Description

As a Pebble user, was thought that high refresh rates to screen eats up battery faster than a low refresh rate to screen. Could the dev of the new Terminal watchface please give users the option to not display the seconds thus extending battery.  

If it is not the case with PineTime and doesn't affect battery life, please delete this feature request.

Avamander, 2022-04-05:
It probably has a very minor impact, but definitely not something as high as Pebbles. Mostly because the screen is not always on, and if it is, the backlight is most of the consumption.

JF002, 2022-08-01:
I really don't think this feature would have any measurable impact on the battery life : refreshing the seconds display happens only once a second (which is an eternity in computer time) only when the display is ON and when the terminal watchface is enabled. 
I consider this as an artistic choice from the author, so, unless I'm wrong about the power usage, I guess we can close this feature request.

closed 2022-08-01 by JF002