InfiniTimeOrg/InfiniTime #1091: Resetting firmware resets to the second last one which was booted

watertrainer, 2022-04-15:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

When updating the watch over the air twice and resetting it after it goes to the second last which was booted and not the last one which was validated

### What should happen instead?

Pressing the reset button should reset the watch to the last validated firmware

### Reproduction steps

The easiest way is just to install the old 1.8.0 dfu using Gadgetbridge, in the firmware setting it shows 1.8.0 and gives the option to reset/validate. 
Now installing the 1.8.0 dfu zip again on top of the old firmware the version doesn't change the version and it still gives the option to validate to reset (note that I never validated a firmware image), if pressing reset though and going back to the firmware setting it still says 1.8.0 and gives the option to reset/validate. 
There is no way to go back to 1.9.0 now without reloading the image using Gadgetbridge.

### More details?

I don't know if this might be an issue in the mcuboot repo or infinitime.

### Version

1.9.0

### Companion app

Gadgetbridge v0.64.0

Riksu9000, 2022-04-15:
This is currently expected behaviour. There are two slots. Primary slot, which is the currently running firmware, and secondary slot, which is the next/previous firmware.

When you downgrade from 1.9.0 to 1.8.0, the 1.8.0 firmware is first placed in the secondary slot, overriding whatever was previously there. Then the bootloader swaps the 1.8.0 to the primary slot and 1.9.0 to secondary slot. If you reset now, the firmwares get swapped and you're back to 1.9.0 in primary slot. Now if you start another firmware update, the new firmware will again override the secondary slot, which currently contains the validated 1.9.0 firmware. The bootloader still swaps them, but now you're left with 1.8.0 in both the primary and secondary slot, with no trace of 1.9.0 left.

I think the firmware update process *could* directly override the primary slot, but if an error occurs, you're left with an unusable watch until the bootloader recovers the firmware from the secondary slot, or another firmware update was started with the help of recovery firmware. Another option is to use up another 464KB of external flash to store the validated firmware, which perhaps isn't very compelling either.

watertrainer, 2022-04-15:
Couldn't the bootloader, if the image in the primary slot didn't get validated and another update process is started, swap the images again and override the, now in secondary unvalidated image, instead of the validated image?

Riksu9000, 2022-04-15:
That would mean that before the firmware upgrade could start, the watch would have to reboot and swap the firmwares. It could work assuming that the validated firmware is never broken and the process is successful. If however there's some issue or difference between firmware versions that causes an incompatibility, the process could fail. When this firmware update process fails, the unvalidated firmware may be partially overridden and unusable. This would mean that an unsuccessful firmware update will in effect revert to the validated firmware. This would also make the process more error prone and complicated, so probably not worth the tradeoff.

In the meantime you can accomplish the same thing by manually reverting to the validated firmware before starting another update, if that's preferred.

closed 2022-04-15 by watertrainer