InfiniTimeOrg/InfiniTime #1096: Quick launces

medeyko, 2022-04-18:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Quick launch for torch, notifications switch, alarm, calendar, bluetooth status, etc.

### Description

Some actions are performed very often, and going through menus is annoying.
Sometimes you need to do an operation quickly.
Sometimes you wish to do an operation not looking at the screen.
All this may be solved with quick launches, which run apps in a predefined state or do something else.

I propose the following quick launch gesture: 
1. Press the hardware button.
2. Keeping the hardware button pressed, do one or more swipes up and down. (Different sequences run different apps/events.)
3. Release the hardware button. The app/action will then be launched/done.
There's even no need to indicate anything before the app is run. However, it would of course be convenient if the name of the app that is going to be run was displayed.
The rationale is that it is quite unlikely that this gesture may be applied unintentionally, yet is it simple enough to do without looking at the screen.
As far as I undertand, it's relatively easy to implement this.

Now for the specific apps/events I propose. I will write only directions of swipes for short. So, "Up+Down" means "press hardware button, then swipe up, then swipe down, then release the hardware button".
1. Up. Turn on flashlight on full light. This is a much needed feature, and indeed the discussion on the flashlight sparkled the idea of this feature. I use the torch often, I want it to be turned on quickly, and I wish to turn it on not looking on the screen. So this is the best candidate for a quick launch.
2. Down. I don't use notifications, but guess that for many it is a regular task to either turn notifications off, or toggle their status, not sure what would be the best approach.
3. Up+Up. Alarm clock app. If possible, set to current time + 8 hours (it's a standard sleep time).
4. Down+Down. Turn the bluetooth off (or switch the status). There are cases when you need to turn bluetooth quickly, for example at a plane. Making the gesture similar to notifications geture makes it easier to memorize it.
5. Up+Down. Calendar app. It is not merged yet, but I believe it will. When you need a calendar, you need it quickly.
6. Down+Up. Calculator app. It is not merget yet to, but probably will. When you need to quickly calculate something, you wish to do it quickly.
There may be even tertiary gestures, like Up+Up+Up swipes, but I don't know which apps or actions for.
Also horizontal Right and Left swiped might be used, but they may be less convenient than vertical Up and Down swipes (because when one presses the hardware button, one also presses on the right side of the watch body too).

Maybe I missed some events that one would like to run often/quickly/not looking on the screen. Please comment!

kieranc, 2022-04-28:
Honestly, pressing the button and doing a gesture on the screen seems like it would require more flexibility than my fingers have to offer, but we do have button double press and long press available, they just need to be user-configurable (currently double press opens notifications and long press opens sysinfo, placeholders really)
Would 2 shortcuts on the button be enough, if they were user selectable?

medeyko, 2022-05-01:
@kieranc have you tried to experiment? Maybe you just didn't find a convenient way to do these gestures.

I found three ways, they all worked fine for me:

1. Press the hardware button with your thumb (with the right side of the watch against the back of your hand), and then swipe with your index finger; OR
2. Press the hardware button with your thumb, holding the right side of the watch with your middle finger, and then swipe with your index finger; OR
3. Press the hardware button with your index finger (with the right side of the watch against the back of your hand) and then swipe with your thumb.

I see no problems. Maybe I would record a video?

As for the user-configurable hardware button double press and long press, I have three points:

1. It is a distinct proposal, not interfering with this one. We can have both quick gestures I proposed AND these user-configurable presses.
2. As far as I know, developers prefer to make Infinitime to minimize configuration options, and to make the Infinifime convenient out-of-the-box.
3. Two shortcuts seem to be too few.

However, I think, as a distinct proposal the user-configurable hardware button double press and long press are really worth considering!

Riksu9000, 2022-07-02:
Clever idea, but I'll close this as the idea isn't aligning with InfiniTime's simplicity principles and the number of actions required to perform is often not lower than when just navigating the normal way.

closed 2022-07-02 by Riksu9000

Shatur, 2022-07-02:
> Clever idea, but I'll close this as the idea isn't aligning with InfiniTime's simplicity principles

But isn't the key audience for the device geeks?

> the number of actions required to perform is often not lower than when just navigating the normal way.

Sadly. Because, for example, I use a flashlight very often and and as for me it requires to much actions (power on, swipe right, touch on flashlight and another touch to enable). On my phone, for example, I simply hold the power button when the phone is locked and it enables flashlight.

Maybe we could allow to configure at least power button? Via single setting with a few predefined actions.

medeyko, 2022-07-03:
> the number of actions required to perform is often not lower than when just navigating the normal way

It is definitely wrong.
1. Light. "`Button` + `swipe up`" vs "`button/raise/shake` + `swipe right` + `precise click` + `imprecise click`" = 2 vs 4
2. Notifications. "`Button` + `swipe down`" vs "`button/raise/shake` + `swipe right` + `precise click`" = 2 vs 3
3. Alarm. "`Button` + `swipe up` + `swipe up`" vs "`button/raise/shake` + `swipe up` + `swipe up` + `precise click`" = 3 vs 4
4. Bluetooth. "`Button` + `swipe down` + `swipe down`" vs "`button/raise/shake` + `swipe down` + `precise click` + `swipe up` + `precise click`" = 3 vs 5
5. (Calendar.) "`Button` + `swipe up` + `swipe down`" vs "`button/raise/shake` + `swipe up` + `swipe up` + `precise click`" = 3 vs 4
6. (Calculator.) "`Button` + `swipe down` + `swipe up`" vs "`button/raise/shake` + `swipe up` + `swipe up` + `swipe up` + `precise click`" 3 vs 5
Each additional movements is problematic for often used actions.

Also not only the number of actions matter, but also their nature too. Please note that all non-shortcut movements include at least one `precise click`. These clicks require visual conrol. Even scrolling requires visual control in order to be certain that you scrolled to the right page.
On the other hand, the actions proposed for the quick launch do not require visual feedback and are not prone to imprecise movements (no misclicks are possible). These are not subtle differences, it's a game changer.

But I see that there's no much interest in this feature, alas.
Ok, maybe it will be different if there is complete pull request.

medeyko, 2022-07-03:
As for the simplicity, it is not necessary to memorize the gestures: while the button is hold, there may be displayed a list of shortcuts. It may be even a list with clickable items, if there are people who prefer precise clicks to imprecise gestures.