InfiniTimeOrg/InfiniTime #1098: Option to rotate screen

5iver, 2022-04-19:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

I'd like the screen rotated 180 degrees, so that the watch could be worn with button on the right

### Description

When I first got my PineTime and tried it on, I instinctively put it on with the button to the right, which caused the screen to be upsidedown when I turned it on! I am right-handed and have become accustomed to having buttons on the right-hand side of things and using the index finger on my right hand to toggle them. To achieve this with a PineTime, an option could be added to rotate the screen 180 degrees. Wearing the watch on the left wrist with the button on the right side would also prevent accidental button presses by sleeves, which I have encountered when wearing a winter coat or leather jacket.

JF002, 2022-04-24:
Someone began working on it [a while ago](https://github.com/InfiniTimeOrg/InfiniTime/pull/98), but it still tagged as Work In Progress. I think we had a few issues with the hardware scrolling feature we use in InfiniTime.

camlafit, 2022-08-27:
Hello

Could be a nice feature, as left writer, it's a bit complicated to use correctly lateral button :)
As pinetime is a square, could be interesting to manage 90° rotation

moriel5, 2022-09-22:
Interesting, for me it feels natural to have the button facing the wrist, and as I too am left handed, that means to the left of the screen, since it is on my right hand (I am used to using my pointer/second finger to press buttons).

If anything, I would prefer to flip other watches that have the button to the right of the screen.