InfiniTimeOrg/InfiniTime #1099: No steps are counted

jeena, 2022-04-20:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

My step counter is always at 0

### What should happen instead?

The step counter should count the steps

### Reproduction steps

I'm not sure how to reproduce it. I can only say what happened. One day I woke up and my PineTime, which I until then had for a couple of weeks and didn't update the software for it, showed me that I had walked 536 bilion steps that morning already.

[image]

I didn't do anything to it and it ran out of battery some day and after that once I charged it and booted it it showed 0 steps, even after walking for a long time with it on my wrist.

[image]

### More details?

I did a factory reset which didn't help. I also Upgraded the firmware to the newest version 1.9.0 and it didn't help either.

There is this gyroscope application on it, and before that happened it showed different colored lines when I was moving the watch. This application does not show anything now anymore either.

https://user-images.githubusercontent.com/151764/164153369-a74d3846-5eec-4d8d-ab70-6a2bdc4819e7.mp4

I'm not sure if this is a hardware error or a software error. It didn't fall down or anything, I just woke up one day with it on my wrist and it was like that.

### Version

1.9.0

### Companion app

Gadgetbridge 0.65.0

alwayslurking, 2022-04-20:
I saw exactly the same symptoms, and found this discussion on the pine forums:

https://forum.pine64.org/showthread.php?tid=16182

Somewhere there unsealed their watch and found a loose chip. I now wonder if an intermittent accelerometer connection was the cause of other random problems I had with my watch, now all resolved at the expense of nothing that uses the accelerometer working.

jeena, 2022-04-21:
Very interesting. I have no other issues but the accelerometer, thanks for the link, good to know that I'm not the only one.

JF002, 2022-04-24:
It looks like InfiniTime cannot initialize the accelerometer or that it's not working at all.
Could you have a look at the 'About' page (last entry in Settings). On the 2nd page, there's a field  'Accel.' What does it show (BMA421, BMA425 or ???).

Anyway, I think this is mostly a hardware issue, and I don't really know if there's anything we can do on the software side to solve that issue :/

sssmack, 2022-04-27:
??? for me too.
It fell and hit the hardwood floor once.

jeena, 2022-04-28:
Yeah, mine shows "???" too, bummer. I guess I'll order a new one then and close this issue because it's clearly not a software issue. Thanks for the explanations!

closed 2022-04-28 by jeena