InfiniTimeOrg/InfiniTime #1105: failed wake up using DoubleTap with active Shake Wake

aemkai, 2022-04-24:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

DoubleTap wakes watch and set it to sleep at once.

### What should happen instead?

DoubleTap should wake watch and watch should stay wake.

### Reproduction steps

1. go to Options - "WakeUp" and enable "DoubleTap" and "Shake Wake"
2. calibrate "Shake Wake" in Options - "Shake Calibration" to about one fourth of maximum
3. set clock to sleep
4. tap double to wake up (not too soft, at best wear the watch)

### More details?

The behavior is only sometimes and according as hard the touches are. For me, calibration is to about one fourth of maximum as suitable for everyday use, and this issue affects two of three wake trials.

The first tap wakes the watch by shake wake, then the second tap starts the doubletap-event and is - since the watch is already wake now - interpreted as set to sleep double tap.

Maybe a short off-time after automatic wake up (e.g. by shake) could avoid this.

### Version

1.9.0

strayobject, 2022-05-19:
I can concur that this is happening to me as well. Since release 1.8 I believe. It is only happening when shake sensitivity is calibrated to a low level. I have mine at a 5th and double tap seems to be picked up as shake wakeup as well. From half onwards double tap works as expected. From 3rd down double tap is broken. 
[image]

It might be a case that at such high sensitivity (or low setting) there are hardware limitation that make it hard to distinguish between shake and a tap, but I will leave that to the experts :)

I would also like to add that with a watch calibrated to a 5th, setting it to shake + single tap works fine even when double tapping.

geekbozu, 2022-05-22:
Thanks for the feedback, There does seem to be a few  points of content on stuff breaking with shake wakes logic for waking up a device, I'm guessing I missed a qualifier when I first got the code working.

I likely need to check for a Tap event and Ignore the shake when that happens as a tap event can look like what triggers a shake. As they both work on a similar mechanism of looking for an impulse noise, Shake Wake adds some additional filtering, however it clearly needs more!

Thanks for the reports they will definitely help with debugging!

doidge-kyle, 2022-10-06:
Also wanted to note that this happens with double-tap to turn the display off. With both shake an double-tap on, if you double-tap to turn off the display it flashes off then comes back on. Usually a second double-tap will turn it off for good (not sure if this is from being more careful or if the second turn on has something to do with it)

Figured this extra scenario may help with testing.