InfiniTimeOrg/InfiniTime #1110: Optional setting to auto launch music control app

Zandengoff, 2022-04-27:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Add optional setting to launch music controls when music starts playing on phone

### Description

Former Pebble user here. One of the main reasons I gravitated towards the Pinetime is the simple no hassle music controls. Works great except for one feature I dearly miss. Music Boss on the pebble contained a setting that upon playing music would switch to the music control app on the watch. This allowed you to simply look down and quickly control your music. As it stands with Infinitime, I have to swipe down and tap the small music app button before controlling music. I know it seems like a small thing, but this is especially problematic while running or biking where launching the app stops the activity and attention is taken away. I know the watch is receiving data from gadgetbridge regarding play state, but I don't know if it is sent even while in a normal watchface or specifically pulled only when the music app is active.

Zandengoff, 2022-04-28:
If it will take a lot of work I can live with button presses for a while, would be a good solution for the short term, especially if already implemented.

Long term though, it would be much slicker to auto launch. It makes sense from a usability perspective, if you are playing music, wouldn't you want the controls? This function could also be used for something like the navigation screen, once direction data gets sent to the watch -> auto launch navigation. (I know navigation has limited use right now, but there is a promising gadgetbridge feature in the final development for compatibility with osmand maps.)

The question I still had, is gadget bridge still sending the data even if on the watchface or does it only get pull when requested by the app? Apologies for my ignorance.

kieranc, 2022-04-30:
I suppose my issue is that I personally couldn't do what you're asking for, it's not to say it can't be done but it's beyond me - if would require that both the phone music app and infinitime be modified to communicate with each other, more than likely that the companion app would need to support this functionality as well, so it's not entirely impossible but it's a lot of work and would have to be taken on by someone more talented than me.

I'm going to have a shot at customising the double press/long press button functions which will hopefully be useful to more users and use-cases than this idea, and I think I can probably make it work ;-) 

I'm not 100% sure I've understood your last question, the companion app pushes notifications to the watch, the watch doesn't request them. The watch can send commands back to the companion app (music controls for example) but the companion app has to know what to do with these commands.

JF002, 2022-04-30:
That's an interesting feature request ! However, since we are trying to limit the number of parameters, would a "music" notification (with quick controls) be a acceptable alternative? I'm thinking about a notification similar to the call notification, for example.

keforbes, 2022-05-06:
I was actually thinking about this the other day.  On the default watch face, the battery icon is in the top-right, the heart monitor is in the bottom-left, and the step counter is in the bottom-right.  So the top-left corner is typically empty.  If music is currently playing on the phone, I wish we could put the music icon in the top-left of the watch face such that if I tapped it, it would jump straight to the music app.  That way, I don't lose the ability to quickly check the time while music is playing, but I can also quickly jump to the music app (without much precision) when I do want to control the music.

keforbes, 2022-05-06:
> would a "music" notification (with quick controls) be a acceptable alternative? I'm thinking about a notification similar to the call notification, for example.

Wouldn't this solution only work if you receive less than 5 of any other notifications (texts, phone calls) while music is playing?  Wouldn't the "music" notification scroll off as new notifications are received?  Or can you "pin" the notification to always be in slot 1/5 while music is playing?  Sorry, I'm not trying to dismiss your suggestion, I'm just asking for clarification.

kieranc, 2022-05-08:
I've been thinking about pinned or priority notifications as a useful feature, for navigation or music, I would assume it would persist and remain on top of the stack if you had this feature enabled

Buddhalobes, 2022-05-12:
I had a thought in this vein and I think it would be better suited here than a new thread.

Would it be possible to have a swipe left to a pinned application. Swipe right is settings, swipe down is notifications, swipe up is app list, but nothing at the moment is on swipe left. the option could be put in setting and maybe makes it watch-face agnostic. I am a service worker and can't program, but having a pinned app to a swipe would be really useful since I want to get to the music app quickly but prevents my environment hitting the controls while I am moving about.

mashuptwice, 2022-07-11:
> I had a thought in this vein and I think it would be better suited here than a new thread.
>
> Would it be possible to have a swipe left to a pinned application. Swipe right is settings, swipe down is [...]

I second this approach. It would certainly be nicer than cluttering the notifications. Maybe it would even be possible to create a setting to let the user decide which app he wants to pin.

mashuptwice, 2022-08-29:
On [my Forks](https://github.com/mashuptwice/infiniTime) development branch I have swipe-left on the watchface set to open the music app, just like the quicksettings:
[image]

The current behavior is a bit weird, as I've kept the music app in the applist and also kept the swipe-down gesture.

Also the animation is not as smooth as with quicksettings. I remember to have read about some trick used, as the animation rolls against the refresh direction, but I couldn't find it anymore. 

 If I find the time, I might break out the individual commits, remove it from the apps and create a clean PR from that.