InfiniTimeOrg/InfiniTime #1114: Build Firmware with Optional Apps

boredom101, 2022-05-02:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Ability to choose which apps are included in the firmware

### Description

I think it would be useful if there was a way to choose which apps are included in your build.
I might try this myself if there is interest.

Jasperki, 2022-05-02:
I would vote for this.
Especialy the possibility to remove some, like pong or paint.

Riksu9000, 2022-05-02:
It's already possible to disable apps in a build simply by editing `ApplicationList.cpp` and replacing `Apps::(appname)` with `Apps::None` and reorganizing the arrays.

Currently InfiniTime includes all apps by default, so I'm guessing the suggestion is that we maintain more apps which wouldn't be in the official build, but can be swapped in during a build @boredom101?

Riksu9000, 2022-05-05:
If you're going to be tinkering with it, then I'm sure you can figure out a way to remove those apps from the build to free up some space, even without this being a supported feature. InfiniTime also isn't limited to 12 app slots. More screens can be added. Just to be clear.

GIJack, 2022-05-28:
this would be great. de-clutter the UI, just use the apps you need/want

mark9064, 2025-06-28:
This has been implemented with -DENABLE_USERAPPS :) See #1894

closed 2025-06-28 by mark9064