InfiniTimeOrg/InfiniTime #1123: Steps/Trip counter maxes out at 99999+ steps

Danimations, 2022-05-10:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

Counter for "trip" stops counting at 99999 steps

### What should happen instead?

Trip counter should include one extra digit, accommodated by slightly adapted layout

### Reproduction steps

Walk 99999+ steps and you'll see that the "trip" meter stops counting.

### More details?

I think the easiest way to improve this would be to make some minor adjustments to the step counter layout. If the number of steps and grey word "steps" were positioned slightly higher, the goal and trip counters could then also be moved further upwards, so that there is room for an extra digit beside "Trip".

Also, the gap between "Goal:" and the corresponding number could be reduced 50% to improve the overall aesthetic. The same could be done to the gap between "Steps:" and its corresponding number.

### Version

v 1.9.0

### Companion app

Gadgetbridge

ItzSwirlz, 2022-10-02:
To be honest I think the best solution is to make it a Scroll widget

kilobyte, 2023-05-12:
While usually a fixed limit is a bad idea, what about keeping it simple and indeed adding that single digit?  For a regular human, 99999 steps is unreachable — we're talking about more than two marathons' distance in a single day.  So Dan managed that — but I don't think anyone can reach a million.  Any takers for this challenge? :grin:

kilobyte, 2023-05-12:
[Here](https://www.menshealth.com/fitness/a36751439/walking-100000-steps-fitness-challenge-24-hours-results/)'s an article about the effort needed to reach six digits.  That's walking, as for running — longest ultramarathons held reach twice that distance; running gaits are far longer than walking though, which means there's less steps taken.

Apparently for sportspeople [steps/time rate is constant](https://en.wikipedia.org/wiki/Running#Good_technique) regardless of speed, at 185-200 steps/minute; it's only stride length that changes, by ~3× between sprint and ultramarathons.  Thus, assuming someone can run 24h without resting, they still won't break past 24×60×200 = 288000 steps.  This gives a comfortable safety factor against someone picking a highly unnatural gait to game the step count.

But then, you need extreme effort for "just" 100000, making this bug a quite low priority...

FintasticMan, 2023-05-13:
I assume that the issue isn't necessarily about the daily step counter stopping at 99999, and more about the trip step counter, which continues counting until you press the reset button. So even though you might not realistically walk that in 1 day, you can walk it over the course of a whole "trip".