InfiniTimeOrg/InfiniTime #1134: Watchface | Heart rate monitor like in hospital

aha999, 2022-05-11:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Watchface as a hospital heart rate monitor that is animated.

### Description

I think it would be cool if there was an option to display your heart rate like in hospital monitors, maybe even as a Watchface (home screen)

Example video: https://youtu.be/yEb4kojtVHw

JF002, 2022-06-06:
Good idea ! This feature already exists in waspos : https://www.youtube.com/watch?v=lIo2-djNR48

ajack2001my, 2022-06-30:
@jf002 will this be ported to the InfiniTime build? When will the watch start running apps from storage? I get the impression everything is now in the firmware and space is kinda tight at the moment.

JF002, 2022-07-08:
@ajack2001my 

> will this be ported to the InfiniTime build?

It really depends if someone is willing to implement it and if the core developers review and merge the changes in the code :)

> When will the watch start running apps from storage? 

The answer is similar to the previous one : when it's ready... Dynamically loadable apps is a very complex topic and will require quite some time from skilled developers to implement!

> I get the impression everything is now in the firmware and space is kinda tight at the moment.

I pay a lot of attention to the memory needs of the changes that are going to be merged, and sometime have to refrain from merging some good changes because of memory usage. My goal is to avoid filling the memory so much that we cannot add anything new to the code. My work on the external memory will probably give a bit of fresh air to the project, though :)

Lillian-Oostrom, 2022-07-27:
It's good to note that this "animation" is an EKG diagram. This shows the electric potential in the heart as it contracts and relaxes. What the pinetime has is a PPG measurement, which only measures the flow of blood through veins (an indirect measurement of the same thing). The real graph this creates is a kind of sine wave, the small "bumps" are not really able to be measured, only guessed at. Showing this is a bit iffy, while it looks cool, we don't want people to think this shows more medical data than it actually is able to. This (mediocre) PPG sensor is nowhere close to the quality of an EKG machine.

mashuptwice, 2022-07-27:
I could imagine this being a huge benefit for tuning the measurements in code!

It would also enable one to judge better if the hear rate reading could be trusted or if there was to much noise. Currently even small movements often lead to a much higher measured heart rate.

joseph58tech, 2022-07-28:
To implement something like the graph from Wasp OS it would take more ram then the developers (and me somewhat) would be comfortable with, it's kind of hard to work with 64k of ram plus trying not the completely fill it, especially with the never ending new implementations we already see on infiniTime. I don't want to be pushy towards Pine64 but I think it will be best if you guys maybe start thinking about a second edition Pinetime, or just making start making pinetime with an upgraded memory chip (Maybe 128k?) we obviously don't need it now, but in a year or so maybe.

ajack2001my, 2022-07-28:
Correct me if I'm wrong here @JF002, but to comment @joseph58tech, the watch's secondary storage is not used much at the moment and everything in the watch right now is all kept in memory regardless if we use it or not.  Once that is sorted, we can store all individual apps and components into storage and only load up whatever app the users wants and there will be more memory for each app to run with. See @JF002 comments 3 posts up! 

I don't think we need a hardware revision of the PineTime just yet when we've hardly scratched the surface of this revision...

joseph58tech, 2022-07-28:
> I don't think we need a hardware revision of the PineTime just yet when we've hardly scratched the surface of this revision...

I know, but I think a few years down the line (assuming infiniTime is still in production (I hope)) we might just need a little bit more ram to keep a seamless experience. Also I can assure the storage will be fine for a very long time (assuming we can get the external storage thing working). But honestly 512k of firmware flash is probably enough at the current rate of storage per implementation.