InfiniTimeOrg/InfiniTime #1135: Make battery reporting more frequent and consistent.

WowieMan, 2022-05-12:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Change battery reporting to make it more frequent and consistent.

### Description

So recently, I’ve been reworking the battery chart in InfiniLink, and because of the limitations of iOS, I've found it would be useful to have a more frequent and consistent stream of battery data. That way InfiniLink can tell when the watch is connected and disconnected in the chart.
Now my idea is to change battery reporting to when the watch checks its battery, that way, InfiniLink should get the battery percent every 10 minutes. Though I'm not sure if this would cause any problems with other companion apps, so I thought I'll ask here.

Riksu9000, 2022-05-13:
Gadgetbridge adds each notified percentage to the chart as a point, so it's a bit nicer when not sending duplicate values constantly, however this isn't a good reason to do things one way or another, since Gadgetbridge could also improve the chart rendering.

So the question is if it's worth sending the same value constantly, and I'm not sure it is. Are you sure there's not another way to tell when the watch is connected?

WowieMan, 2022-05-13:
Looking a bit more into it, I think I can log disconnects in InfiniLink, so I'll close this feature request.

closed 2022-05-13 by WowieMan