InfiniTimeOrg/InfiniTime #1141: Battery indicator bug

aha999, 2022-05-17:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

At the end of the first charge of the new pinetime i checked the battery status and it said 100%. Then i disconnected the pinetime from the charging dock and immediately reconnected it, so it started charging again, but this time it said that the battery is at 98%, although it said 100% 2 seconds ago before i unplugged it. The second time i did that, the  battery indicator immediately jumped from 100% to 96% at the second of unplugging it.

### What should happen instead?

If the battery was actually charged at 100%, it shouldn't display 98% after immediately unplugging and replugging it back... It's impossible to lose 2% in 2 seconds. The battery should still show 100% if it was actually charged at 100%.

### Reproduction steps

- Buy the pinetime
- Charge the device for the first time
- Check the battery status in the settings when it is at 100%, 
- Disconnect the PineTime from the charging cradle/dock
- Reconnect the PineTime back to the charging dock 2 seconds later/immediately
- The battery status will display 98% or 96% charge after unplugging it (in my case of 2 tries)

### More details?

One theory is, that It is possible that i disconnected it exactly at the point when it reached 100% so disconnecting it jumped back to 98%... which is a coincidence.

But when i did this the second time, it immediately jumped to 96% when disconnecting it.

### Version

1.6.0

Riksu9000, 2022-05-17:
The behaviour is different in newer InfiniTime versions. Please update the firmware.

closed 2022-05-17 by Riksu9000

escoand, 2022-05-19:
Maybe this comment also helps to understand this issue: https://github.com/InfiniTimeOrg/InfiniTime/issues/804#issuecomment-1120151015