InfiniTimeOrg/InfiniTime #1143: Screen becomes unresponsive when playing table tennis game

aha999, 2022-05-18:
### Verification

- [X] I searched for similar bug reports and found none was relevant.

### What happened?

When i was playing the built in table tennis game, she screen stopped responding a little bit in the game, and my finger was on the screen for the entire time.

### What should happen instead?

If you're playing the game and touching the screen, the touch shouldn't suddenly become unresponsive.

### Reproduction steps

1. Buy PineTime
2. Install currently latest Infinitime 1.9.0, i have the device connected to the watch
3. Open the table tennis built in game by swiping up on the homescreen, then swiping up again, then clicking on a _table tenis racket with a ball_, then playing game for some time and holding the finger on the touch responsive parts of the PineTime.
4. The screen will stop responding
5. release the finger and tap again to get responsivness back

### Version

1.9.0

### Companion app

Gadgetbridge

JF002, 2022-05-18:
The touch controller has an auto-sleep functionality which might be the cause if this behavior. I don't know if this feature can be disabled to avoid that kind of behavior in use-cases that need the touch controller to work continuously.

ajack2001my, 2022-05-19:
> The touch controller has an auto-sleep functionality which might be the cause if this behavior. I don't know if this feature can be disabled to avoid that kind of behavior in use-cases that need the touch controller to work continuously.

Shouldn't the auto-sleep see that there is activity/input from the sensor and not auto-sleep?