InfiniTimeOrg/InfiniTime #1146: Pop-up validation window after updating Infinitime firmware

aha999, 2022-05-18:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Update validation info pop-up after updating the Infinitime firmware

### Description

It would be more user friendly if after updating the firmware on the pinetime, a window would pop-up, saying that if you want to keep the changes/update to not be reverted, you have to validate the update after you test it. This would prevent the unexpected update reversal.

https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/doc/gettingStarted/updating-software.md#firmware-validation=

ajack2001my, 2022-05-19:
> ### Description
> 
> It would be more user friendly if after updating the firmware on the pinetime, a window would pop-up, saying that if you want to keep the changes/update to not be reverted, you have to validate the update after you test it. This would prevent the unexpected update reversal.
> 

How will the watch know you've tested everything to make sure firmware works correctly? That's why it's manual, so users can test anything and everything to their satisfaction before verifying and is able to simply restart watch to revert to previous firmware.

Hope my explanation is easy to understand... and that I understood this function correctly in the first place... :)

Avamander, 2022-05-19:
This has been discussed to some extent earlier. Making it too quick and easy has the potential downside that people validate broken firmware.

aha999, 2022-05-19:
I think that they should be at least reminded at first start, to test the Pinetime for a certain amount of time before applying changes... which would remind them of this requirement if they don't want the update reversed unexpectedly. I didn't know what that feature was until i specifically searched for it.

ajack2001my, 2023-02-21:
What about prompting user if a reboot was detected without firmware validation? Keeps asking at each reboot until validated.... That was people only validate when they think it's good... Just my $0.02

Riksu9000, 2023-02-21:
I'm not sure what you mean. Rebooting without validating will revert the firmware, and I don't think rebooting is common practice nowadays.

ajack2001my, 2023-02-21:
@Riksu9000 prompt before reverting...

Riksu9000, 2023-02-21:
The reverting is a safety measure against bricking and is assumed to only be used when there's a problem with the firmware. Adding a prompt before that could add another point of failure. It's still an interesting idea, but would require re-imagining the entire reverting feature, which is not in the scope of this issue.

isaacc27, 2023-03-21:
I'd like to work on this if no one else has taken this up. Can I get confirmation that the implementation is checking if the firmware has been validated after some time (an hour), and then if not send a reminder notification?

Riksu9000, 2023-03-22:
Yes, I think that is the best way to implement this currently. I would implement it using a FreeRTOS timer in DisplayApp.

Riksu9000, 2023-03-30:
It should definitely be preview, because it's the same as any other notification. If you push to notificationManger and push a new notification message to SystemTask, it will open a notification preview. Are you doing something different?

A button might be worth experimenting with, but let's keep the changes per PR to a minimum.

ic-27, 2023-03-30:
No it's nothing, I was just experimenting with Notification, as it doesn't seem to be used anywhere.

timaw7, 2024-07-17:
Just got burned by this myself because I was unaware of the validation button.  It seems like the difference between a test system and production system.  If i'm flashing an update, I would expect it has been quality checked and validated by others beforehand.