InfiniTimeOrg/InfiniTime #1147: Bluetooth settings / add Discoverable button

aha999, 2022-05-19:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Pinetime's bluetooth is only visible for a short amount of time after restart. Add a "visible for 30 sec." button to Bluetooth settings inside infinitime to solve this issue.

### Description

Currently, we have to restart the Pinetime if we want to make it visible to other devices. This behaviour is confusing and non-obvious, so people often ask how to connect to pinetime when it is not discoverable.

I suggest that you add a "Discoverable for 30 seconds" button to the Infinitime Bluetooth settings, which would make pinetime's bluetooth beacon/connectivity discoverable for a set amount of time. 
It could also become immediately undiscoverable when you exit bluetooth settings manually if you didn't connect while it was still discoverable in that period of time. 

It also makes it much more convenient to do it this way and makes more sense that this setting would be available there.

Avamander, 2022-05-19:
Are you on latest InfiniTime?

aha999, 2022-05-19:
Yes. Bootloader 1.0.0, Infinitime 1.9.0.

Riksu9000, 2022-06-20:
The thing is, InfiniTime is nowadays always discoverable when it is not connected to a device. This feature can be considered, but the current premise is faulty.

closed 2023-02-22 by Riksu9000