InfiniTimeOrg/InfiniTime #1152: Find my phone

WhyNotHugo, 2022-05-19:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Implement "Find my phone": make my phone ring so I can find it.

### Description

I often leave my phone lying around in random places. It's usually the same 2-3 spots, but sometimes it's some super weird location and it takes me a long time to find it.

I'd like to tap a button on my watch and make my phone ring so I can find it. There's actually a Bluetooth profile for this: https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=239389

I think it would be wise to implement the standard, so rather than require phones to have specific Infinitime support for this, they just need to support the standard.

Also relevant: https://www.bluetooth.com/specifications/specs/find-me-profile-1-0/

Avamander, 2022-05-19:
Duplicate of #343

closed 2022-05-19 by Avamander

WhyNotHugo, 2022-05-20:
Oh, I must have made a typo in my search, because that did definitely not show up in the three results I had when searching. :sweat_smile: