InfiniTimeOrg/InfiniTime #1162: Update alarm icon

calhix, 2022-05-29:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

Update the alarm app icon with two bells so it doesn't look like a clock

### Description

Thought this would be an easy change. It would be nice to update the current icon for the alarm from just a clock looking icon to one with the two bell shapes on the top to make it look more like an old school alarm clock and one you see commonly in other alarm apps for one's phone.

calhix, 2022-05-31:
Actually this should be marked as an enhancement.

christianfl, 2022-06-01:
A corresponding fontawesome icon is present: https://fontawesome.com/icons/alarm-clock?s=solid

Afaik these are used for InfiniTime.

Riksu9000, 2022-06-02:
This icon is part of the pro version of Font Awesome, so unfortunately we cannot use it. We'll need to find another icon pack, or create icons specifically for InfiniTime.

christianfl, 2022-06-02:
Ah sorry, missed that.

https://fonts.google.com/icons?selected=Material%20Symbols%20Outlined%3Aalarm%3AFILL%400%3Bwght%40400%3BGRAD%400%3Bopsz%4048

could be an alternative but it would maybe look strange to just use one from another icon pack. Especially because they only have an outlined alarm icon.

calhix, 2022-06-04:
Also found some options here: https://iconscout.com/icons/alarm?price=free
Or we could use the bell icon from font awesome here: https://fontawesome.com/icons/bell?s=solid

mabuch, 2022-06-05:
There is a single bell icon in the free pack that could be used:
https://fontawesome.com/icons/bell?s=solid  (there is also a bell-slash variant in the free font: https://fontawesome.com/icons/bell-slash?s=solid)

PR #1996 (2024-02-12): Alarm: Replace clock icon with bell
Fixes #1162 

[image]
closed 2024-02-12 by FintasticMan