InfiniTimeOrg/InfiniTime #1168: World Clock

calhix, 2022-06-04:
### Verification

- [X] I searched for similar feature request and found none was relevant.

### Pitch us your idea!

An app to to be able to select multiple time zones to track

### Description

I think an app for a world clock would be really useful. I have family that live in different countries that I contact on a regular basis and being able to see their times from the watch would be really useful. Perhaps you could have a list of major cities in each timezone or the UTC times you could pick from to track.

medeyko, 2022-07-03:
I missed the multiple timezones display a lot, but I guess that there are too few memory left on the device to include this feature...

callingmybluff, 2022-07-03:
Can we then please close tickets that can not be added?

FintasticMan, 2022-07-03:
They will be able to be added at some point, when it's possible to load apps from the external flash. There is some really good progress on that from JF, who's got images and fonts to load at a pretty good speed.

calhix, 2022-07-10:
Good work so far, something I was thinking of is akin to the style of the world clock on the apple watch. It would be nice for the first screen in the app to be the list of cities you have selected to track (similar to the apple watch image from the link below). You would then maybe swipe left to a second screen which would have a list of major cities within each time zone (perhaps this would be a good start but it's missing quite a few cities imo: https://i.redd.it/imqp9i619a421.png) with a box to check beside each (just like the Wake Up app).
Perhaps optimally we would have a world clock integrated with gadgetbridge, where you could select the cities you want to track from there to then display within the world clock app on the pinetime.

Here is the icon we can use for it: https://fontawesome.com/icons/globe?s=solid
Apple watch: https://help.apple.com/assets/620FD4265AC6AE3F090A11D6/620FD42A5AC6AE3F090A11EE/en_US/ed82705883009a57983264270a0b94d0.png